PX4/PX4-Autopilot #7911: Logging fails on current master

sanderux, 2017-09-04:
On a DeltaQuad config:

```
INFO  [mavlink] mode: Config, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log
ERROR [uorb] mavlink_log advertise failed
ERROR [logger] Failed to register ULog file to the hardfault handler (-24)
ERROR [logger] Can't open log file /fs/microsd/log/sess004/log001.ulg, errno: 24
INFO  [init] Mixer: /etc/mixers/deltaquad.main.mix on /dev/pwm_output0 
```
ping @dagar @LorenzMeier @davids5 @bresch

dagar, 2017-09-04:
We need to do a pass and get the logger topics list under control.
https://github.com/PX4/Firmware/pull/7559

I think we should have a much more limited core set and expand the logger profiles idea.

dagar, 2017-09-04:
We should limit the number of add_topics() based on CONFIG_NFILE_DESCRIPTORS or have logger handle this situation a little more gracefully. Users are able to change the topic list with params or adding a list to the sd card.

dagar, 2017-09-04:
I'll open a PR shortly, but so far I'd propose.

 - New profile for extended GPS logging
   - gps_dump, satellite_info

 - Replace rc_channels and input_rc with manual_control_setpoint. It's much more useful for understanding what the controllers are doing and logs QGC gamepad input.

 - Remove debug_key_value, debug_value, debug_vect or add to another profile.

 - New profile for things like att_pos_mocap, vehicle_vision_attitude, vehicle_vision_position?

 - Remove task_stack_info? I don't think it's used.

 - Drop differential_pressure logging by default (could be yet another profile).

 - We could get into differenent sets for FW/MC/VTOL if needed.

LorenzMeier, 2017-09-04:
We should drop the debug vect stuff by default. GPS as well. We need the task stack info (check out http://review.px4.io)

In the old logger I dropped stdin and stdout right before opening the log file. That makes it gracefully. We can log a warning as well.

TSC21, 2017-09-04:
I like the idea of profiles more than just dropping the log topics.

dagar, 2017-09-04:
I don't see task_stack_info being used by flight review. Can you confirm @bkueng?

Usually I pick up the stack warnings from console messages (via flightreview).

commit 49dceb2b (2017-09-05): logger update topics (#7914)

 - fixes #7911
closed 2017-09-05 by dagar