PX4/PX4-Autopilot #7916: standart_vtol transition back groudspeed slow down

MikeWellC, 2017-09-05:
HI：
here is my log
https://logs.px4.io/plot_app?log=193d957a-3f5f-4147-bcb5-55056c80a5ce

during vtol transition back period ,attitude pitch setpoint be set to zero,but why pitch seems to be set?

below is flightplot screenshot

[image]

**Bug Report** or **Feature Request**

For general questions please use [PX4 Discuss](http://discuss.px4.io/) or [Slack](slack.px4.io). Issue reports need to contain the items below:

 - Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/en/flight-reporting.html))
 - Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/)
 - Expected behavior and actual behavior.
 - Steps to reproduce the problem.
 - Specifications like the version of the project, operating system, or hardware.

bresch, 2017-09-05:
Hi, it's not a setpoint issue. the controller reacts against this motion, it means that this is due to a disturbance. You can also see that at the same moment, you also have a roll error. Those disturbances are most likely induced by:

1. a wind gust: standard vtols are easily disturbed by wind
2. wing stall: at low speed, a wing can stall before the other, this produces a sudden torque

[image]

I would also suggest you to reduce the back transition time `VT_B_TRANS_DUR` from 8 to 4 or 5 seconds, this will be more than enough for your plane.

MikeWellC, 2017-09-07:
Hi,thankyou,Should be perturbed. Airspeed is low，In order to keep the alttitude, pitch is positive.
but VT_B_TRANS_DUR can not be reduced,the above log is the simulation.The real flight, after back transition,if groundspeed does not  Arrival MPC_XY_CRUISE（default 5m/s),pitch_setpoint will be set to MPC_TILTMAX_AIR.   resulting in that The action is too big, looks insecure, although I know the vtol is no problem.like this
[image]

bresch, 2017-09-25:
I'm still not clear about why this appears in SITL. I never checked what is truly modeled in simulation (wind, airfoils dynamics, ...) and thus I can't tell if it's a bug or not.

closed 2018-02-21 by PX4BuildBot