PX4/PX4-Autopilot #7925: VTOL Standard - Back transition velocity setpoint reset

bresch, 2017-09-05:
During a back-transition, the Multicopter Position Controller continuously resets its velocity setpoint to the actual ground speed, constrained by the maximum allowed velocity in MC mode 
https://github.com/PX4/Firmware/blob/0c38f89b23215713c39987b215025cd6ad753bff/src/modules/mc_pos_control/mc_pos_control_main.cpp#L2244-L2251
However, I saw in a recent log that this value is sometimes reset to zero:
[image]

@sanderux @dagar

MikeWellC, 2017-09-07:
HI,I also saw it
[image]

RomanBapst, 2018-02-13:
@Stifael @MaEtUgR Do you have any idea why this happens?

sanderux, 2018-02-13:
@bresch was this mission mode? it might do that when overshooting the position setpoint
By the way, i think _vehicle_status.is_rotary_wing is true during backtransition so the code you reference would not be running.

Stifael, 2018-02-14:
@RomanBapst no. But from the plot is does not look like that the septoint for speed is continuously reset.

closed 2019-02-12 by stale[bot]