PX4/PX4-Autopilot #7931: Pixhawk X2.1 : No Voltage, nor Amp reading in the QGC.

tubeme, 2017-09-07:
The battery voltage level and amp reading stays at -1 (N/A) for x2.1 hardware.

I get reading from the sensor in the scale of 0-3.3v for the voltage.

I connected the sensor to FMUv2 and everything is perfect.

Tested QGC 3.2.2 and 3.2.4. (at first thinking it is QGC issue https://github.com/mavlink/qgroundcontrol/issues/5621)

@DonLakeFlyer explained that this could not be a QGC issue.

Also tested PX4 1.6.1 and 1.6.5 with the same results.

Checked the wiring at least 5 times.

Tried two versions of the hardware. One original A-UAV and one developed by us.

All with the same results.

Are we missing something?

Thanks

tubeme, 2017-09-08:
With this version:

Hardware Version | PX4FMU_V2
Firmware Version | 50ecee9df817ebc414d606b5b551aab08d86b598

it worked. I can see it in the logs.

Sadly I cannot load this version anymore (so cannot double check), because the boot-loader changed and I cannot flash the regular FMUv2 version to the X2.1 hardware.

LorenzMeier, 2017-09-09:
Hey! I'm traveling right now but will follow up Monday.

tubeme, 2017-09-21:
@LorenzMeier Any ideas why this might happen?

The problem persist we could not find anyway we could enable the voltage and C reading.

davids5, 2017-09-21:
@tubeme - 
Just so I am sure I understand your setup. Do you have a v2.1 HW AKA Cube? 

Are you connecting to power 1 or power 2 or both?

On the Cube px4fmu-v3_default build must be used.

What happens if you use master px4fmu-v3_default on the v2.1 HW?

SinigerVTL, 2017-09-21:
@davids5  I'm using the auav X2.1 it is not the cube. it is auav x2.1 firmware that I'm using.

tubeme, 2017-09-21:
@davids5 I'm using the auav X2.1 it is not the cube. it is auav x2.1 firmware that I'm using.

davids5, 2017-09-21:
Sorry! I am glad I asked!

what does 

`listener system_power` , `listener battery_status` and `adc test`

return?

tubeme, 2017-09-21:
nsh> listener system_power
TOPIC: system_power instance 0 #1
timestamp: 24028485
voltage5V_v:   5.2255
usb_connected: 0
brick_valid: 1
servo_valid: 1
periph_5V_OC: 0
hipower_5V_OC: 0

nsh> listener battery_status
TOPIC: battery_status instance 0 #1
timestamp: 0
voltage_v:   0.0000
voltage_filtered_v:   0.0000
current_a:   0.0000
current_filtered_a:   0.0000
discharged_mah:   0.0000
remaining:   0.0000
scale:   0.0000
cell_count: 0
connected: False
warning: 0

tubeme, 2017-09-21:
nsh> adc test
2: 1503  3: 0  4: 3246  13: 2445  14: 2171  15: 1  16: 6
2: 1498  3: 0  4: 3246  13: 2421  14: 2174  15: 1  16: 6
2: 1508  3: 0  4: 3237  13: 2422  14: 2167  15: 1  16: 6
2: 1498  3: 0  4: 3231  13: 2415  14: 2173  15: 2  16: 6
2: 1498  3: 0  4: 3246  13: 2422  14: 2180  15: 2  16: 6
2: 1499  3: 0  4: 3248  13: 2424  14: 2149  15: 2  16: 6
2: 1502  3: 0  4: 3251  13: 2424  14: 2166  15: 1  16: 6
2: 1503  3: 0  4: 3250  13: 2441  14: 2168  15: 1  16: 8
2: 1498  3: 0  4: 3238  13: 2446  14: 2165  15: 2  16: 7
2: 1498  3: 0  4: 3251  13: 2444  14: 2148  15: 2  16: 6
2: 1504  3: 0  4: 3239  13: 2445  14: 2158  15: 2  16: 6
2: 1497  3: 0  4: 3247  13: 2445  14: 2183  15: 1  16: 6
2: 1503  3: 0  4: 3249  13: 2442  14: 2170  15: 2  16: 7
2: 1500  3: 0  4: 3246  13: 2445  14: 2153  15: 1  16: 6
2: 1497  3: 0  4: 3245  13: 2440  14: 2144  15: 2  16: 6
2: 1499  3: 0  4: 3242  13: 2426  14: 2174  15: 1  16: 6
2: 1505  3: 0  4: 3247  13: 2431  14: 2181  15: 1  16: 6
2: 1490  3: 0  4: 3239  13: 2423  14: 2170  15: 1  16: 6
2: 1498  3: 0  4: 3247  13: 2433  14: 2152  15: 2  16: 6
2: 1493  3: 0  4: 3234  13: 2437  14: 2171  15: 2  16: 6
2: 1501  3: 0  4: 3236  13: 2442  14: 2162  15: 1  16: 7
2: 1498  3: 0  4: 3242  13: 2448  14: 2164  15: 2  16: 6
2: 1494  3: 0  4: 3234  13: 2446  14: 2177  15: 1  16: 7
2: 1497  3: 0  4: 3239  13: 2431  14: 2185  15: 2  16: 7
2: 1501  3: 0  4: 3247  13: 2446  14: 2157  15: 1  16: 6
2: 1503  3: 0  4: 3236  13: 2428  14: 2180  15: 2  16: 6
2: 1494  3: 0  4: 3239  13: 2446  14: 2172  15: 2  16: 6
2: 1503  3: 0  4: 3245  13: 2422  14: 2169  15: 1  16: 6
2: 1496  3: 0  4: 3239  13: 2428  14: 2181  15: 1  16: 6
2: 1500  3: 0  4: 3241  13: 2448  14: 2147  15: 0  16: 7
2: 1504  3: 0  4: 3248  13: 2423  14: 2183  15: 1  16: 6
2: 1503  3: 0  4: 3250  13: 2436  14: 2139  15: 2  16: 6
2: 1503  3: 0  4: 3245  13: 2426  14: 2161  15: 1  16: 6
2: 1503  3: 0  4: 3236  13: 2427  14: 2175  15: 2  16: 6
2: 1503  3: 0  4: 3234  13: 2447  14: 2170  15: 2  16: 7
2: 1498  3: 0  4: 3244  13: 2426  14: 2182  15: 1  16: 5
2: 1502  3: 0  4: 3247  13: 2449  14: 2177  15: 2  16: 6
2: 1496  3: 0  4: 3248  13: 2428  14: 2158  15: 2  16: 6
2: 1498  3: 0  4: 3235  13: 2427  14: 2162  15: 2  16: 6
2: 1500  3: 0  4: 3252  13: 2422  14: 2158  15: 2  16: 6
2: 1496  3: 0  4: 3243  13: 2425  14: 2179  15: 2  16: 6
2: 1505  3: 0  4: 3235  13: 2442  14: 2147  15: 2  16: 6
2: 1498  3: 0  4: 3242  13: 2426  14: 2154  15: 1  16: 5
2: 1502  3: 0  4: 3236  13: 2440  14: 2132  15: 1  16: 6
2: 1501  3: 0  4: 3247  13: 2434  14: 2156  15: 1  16: 6
2: 1502  3: 0  4: 3254  13: 2456  14: 2160  15: 2  16: 6
2: 1499  3: 0  4: 3248  13: 2450  14: 2155  15: 2  16: 7
2: 1499  3: 0  4: 3250  13: 2437  14: 2168  15: 1  16: 5
2: 1500  3: 0  4: 3236  13: 2422  14: 2183  15: 2  16: 7
2: 1497  3: 0  4: 3237  13: 2428  14: 2178  15: 1  16: 6
nsh>

tubeme, 2017-09-25:
@davids5 Any Ideas?

davids5, 2017-09-25:
Have a look in sensors
The DN of chan 2 is 1499/4096 = 1.2 Volts. The scale should be 10 X so the input is ~12 V.

see https://github.com/PX4/Firmware/blob/0780e430fda5396a6ab64380380a993afabb2015/src/modules/sensors/sensors.cpp#L533

Any of those _parameters set wrong?

davids5, 2017-09-25:
@tubeme - if the parameters are wrong, the caling would be off and this line
https://github.com/PX4/Firmware/blob/0780e430fda5396a6ab64380380a993afabb2015/src/modules/sensors/sensors.cpp#L549 being false would give you the results you are seeing

may

tubeme, 2017-09-25:
@davids5 You are correct! Thanks.

After all it turns out that it is QGroundControl Problem... @DonLakeFlyer 

With the Power Menu of QGC the voltage devider was 0.00000 by default and once i wanted to make a Calculate for new voltage devider it did not allow me to remember the number I entered. So I had to forcefully go in the Parameters editor and edit Battery Calibration BAT_V_DIV. Then I saw my voltage in the Title bar. And also I was able to make a Bat recalibration in the Power menu. I have Amp devider at 0.000000 as well. I probably have to make the same procedure...

closed 2017-10-05 by LorenzMeier