PX4/PX4-Autopilot #7937: Error in Multi-uav SITL:How to solve “xacro.py: error: expected exactly one input file as argument”

david-spc, 2017-09-08:
Hello everyone.

I had followed the guide on multi-vehicle simulation: https://dev.px4.io/en/simulation/multi-vehicle-simulation.html in ubuntu 16.04 i686.

But I failed when I run the command that "roslaunch px4 multi_uav_mavros_sitl.launch"

And the output is the following:

Done checking log file disk usage. Usage is <1GB.
Usage: xacro.py [options]

xacro.py: error: expected exactly one input file as argument
while processing /home/burobotics/Guang/src/Firmware/launch/single_vehcile_spawn.launch:
Invalid tag: Cannot load command parameter [iris_1_sdf]: command [/opt/ros/kinetic/share/xacro/xacro.py /home/burobotics/Guang/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.xacro rotors_description_dir:=/home/burobotics/Guang/src/Firmware/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=14560 > iris_1.urdf ; 'gz sdf -p iris_1.urdf'] returned with code [2].

Param xml is 
The traceback for the exception was written to the log file

There exists a same issue in github, but the author closed the issue but have no solution.

Stoven-Dubois, 2017-10-05:
Same thing here.

I followed the tutorial here: https://dev.px4.io/en/simulation/multi-vehicle-simulation.html

So I am doing this in a console:

make posix_sitl_default gazebo 

And in a second console

source Tools/setup_gazebo.bash $(pwd) $(pwd)/build_posix_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 multi_uav_mavros_sitl.launch

I get this on the second console:

Usage: xacro.py [options] <input>

xacro.py: error: expected exactly one input file as argument
while processing /home/alexis/catkin_ws/src/Firmware/launch/single_vehcile_spawn.launch:
Invalid <param> tag: Cannot load command parameter [iris_1_sdf]: command [/opt/ros/kinetic/share/xacro/xacro.py /home/alexis/catkin_ws/src/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.xacro rotors_description_dir:=/home/alexis/catkin_ws/src/Firmware/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=14560 > iris_1.urdf ; 'gz sdf -p iris_1.urdf'] returned with code [2]. 

Param xml is <param command="$(arg cmd)" name="$(arg vehicle)_$(arg ID)_sdf"/>
The traceback for the exception was written to the log file

dronecrewkari, 2017-11-08:
Same here. I have encounter the same problem

Ahrovan, 2017-12-25:
hmm. same problem

hamdanAli, 2018-01-08:
i have the same problem, any solution guys?

YCH188, 2018-02-04:
emm，same here. So disturbing

YCH188, 2018-02-04:
I found a solution,but haven't tried.Maybe it will help.
PX4/sitl_gazebo#101

Ahrovan, 2018-02-04:
I solved this problem with another way

david-spc, 2018-02-04:
@Ahrovan hello, could share the solution on how to do this ?

Ahrovan, 2018-02-04:
1- Go to the folder where you cloned the PX4 Frimware.
2- Go to the launch folder Fimrware_clone/launch
3- open the single_vehcile_spawn.launch launch file
4- Replace this line

 <arg name="cmd" default="$(find xacro)/xacro.py $(find px4)/Tools/sitl_gazebo/models/rotors_description/urdf/$(arg vehicle)_base.xacro rotors_description_dir:=$(find px4)/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=$(arg mavlink_udp_port) > $(arg vehicle)_$(arg ID).urdf ; 'gz sdf -p $(arg vehicle)_$(arg ID).urdf'" />
by

<arg name="cmd" default="$(find xacro)/xacro.py $(find px4)/Tools/sitl_gazebo/models/rotors_description/urdf/$(arg vehicle)_base.xacro rotors_description_dir:=$(find px4)/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=$(arg mavlink_udp_port)" />
5- Try to launch the simulation again,

roslaunch px4 multi_uav_mavros_sitl.launch

hamdanAli, 2018-02-04:
it worked but you have to type the rest of the commands manually, ie:
$(arg vehicle)_$(arg ID).urdf ; 'gz sdf -p $(arg vehicle)_$(arg ID).urdf'"

YCH188, 2018-02-04:
@Ahrovan @hamdanAli Thanks for your solution,but I found a new error.It shows "ERROR [mavlink] [timesync] Hard setting offset."

Ahrovan, 2018-02-04:
Yes, I also saw this problem. Let's take part in solving this problem

Ahrovan, 2018-02-04:
@hamdanAli Explain what you mean

hamdanAli, 2018-02-05:
I think the problem is generating the sdf file as described in the multi-vehicle tutorial [](https://dev.px4.io/en/simulation/multi-vehicle-simulation.html)
it says: "  

> The model's xacro file will be used to generate an sdf model that contains UDP port that you select

"

here is where the xcaro file fails ; in the process of generatong the sdf.
so what I did is:
1- I removed the part that generate the sdf file from the single_vehcile_spawn.launch
ie: replace this code in the single_vehcile_spawn.launch:

`<arg name="cmd" default="$(find xacro)/xacro.py $(find px4)/Tools/sitl_gazebo/models/rotors_description/urdf/$(arg vehicle)_base.xacro rotors_description_dir:=$(find px4)/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=$(arg mavlink_udp_port) > $(arg vehicle)_$(arg ID).urdf ; 'gz sdf -p $(arg vehicle)_$(arg ID).urdf'" />`
with this : 

`<arg name="cmd" default="$(find xacro)/xacro.py $(find px4)/Tools/sitl_gazebo/models/rotors_description/urdf/$(arg vehicle)_base.xacro" />`

2- open a terminal and excute xcaro again with full discribtion of model and port:

`xxx@xxx:~$ xacro [Firmware path]/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.xacro rotors_description_dir:=[Firmware path]/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=14560 > iris_1.urdf`

3-generatre an urdf file which must be converted to sdf by:
`gz sdf -p iris_1.urd
`

then launch the simulation with the commands as in the tutorial:
```
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build_posix_sitl_default
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd)/Tools/sitl_gazebo
roslaunch px4 multi_uav_mavros_sitl.launch
```

lamping7, 2018-02-05:
> 3-generatre an urdf file which must be converted to sdf by:

This step is not necessary.

You can spawn a urdf model directly. I'll fix this and the original error shortly.

closed 2018-02-07 by LorenzMeier

david-spc, 2018-02-27:
@YCH188 @Ahrovan The error "ERROR [mavlink] [timesync] Hard setting offset." is because of the low power of your PC referred to "https://github.com/mavlink/mavros/issues/808"

weiminshen99, 2018-03-24:
Hi, @lamping7,

I am running the latest px4 version for this, but still run into the following error, please help.

~/GitHub/Firmware$ roslaunch px4 multi_uav_mavros_sitl.launch
... logging to /home/prl/.ros/log/65ef6a90-2f5f-11e8-8341-002710aea30c/roslaunch-PRL-13700.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

while processing /home/prl/GitHub/Firmware/launch/single_vehicle_spawn.launch:
Invalid <arg> tag: xacro
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/opt/ros/kinetic/share
ROS path [2]=/home/prl/GitHub/Firmware
ROS path [3]=/home/prl/GitHub/Firmware/Tools/sitl_gazebo
ROS path [4]=/home/prl/GitHub/Firmware
ROS path [5]=/home/prl/GitHub/Firmware/Tools/sitl_gazebo. 

Arg xml is <arg default="$(find xacro)/xacro $(find px4)/Tools/sitl_gazebo/models/rotors_description/urdf/$(arg vehicle)_base.xacro rotors_description_dir:=$(find px4)/Tools/sitl_gazebo/models/rotors_description mavlink_udp_port:=$(arg mavlink_udp_port) --inorder" name="cmd"/>
The traceback for the exception was written to the log file

lamping7, 2018-03-24:
Do you have the xacro package? i.e. `sudo apt install ros-kinetic-xacro`

weiminshen99, 2018-03-24:
Thanks and I forgot that. It is working now.
Alternatively, do you know if we can run multiple SITLs without roslaunch? I am currently using multiple-sitl in #8512 and it has a nice option for "--nomavros". It requires, however, its own sitl_gazebo and I don't know how to make it work with the sitl_gazebo in Firmware/Tools yet. Thanks!