PX4/PX4-Autopilot #7938: Copter falls on changing MPC_Z_VEL_MAX_UP and MPC_Z_VEL_MAX_DN in-flight

okalachev, 2017-09-08:
Hello.

We're controlling the copter in OFFBOARD mode using `SET_POSITION_TARGET_LOCAL_NED`, with position settings. Also, we want to dynamically change maximum velocities of flying.

For that, we use `MPC_XY_VEL_MAX`, `MPC_Z_VEL_MAX_DN`, `MPC_Z_VEL_MAX_UP` settings. Changing `MPC_XY_VEL_MAX` works OK, but additional changing `MPC_Z_VEL_MAX_DN` or `MPC_Z_VEL_MAX_UP` sometimes makes the copter stop the motors and fall.

We have observed this situation several times.

Here's the altitude graph of such falling:

[image]

We're taking off, set parameters, then logger hangs for a while (the straight line), then the copter stops the motors and falls.

Link to this log: https://logs.px4.io/plot_app?log=baa165fc-9aaf-484b-9073-08a1521adb5b.

I can attach some additional logs and video files, if they are needed.

bkueng, 2017-09-11:
What most likely was happening, was that important flight modules were blocked on reading parameters because a parameter save was in progress. This normally does not occur because parameters are not written often.
This helps: https://github.com/PX4/Firmware/pull/7950. It makes sure that the flight modules stay responsive during such operations. However it's still possible that you lose offboard control for several 100ms if a parameter write happens during a saving operation.
I suggest you set the params before the flight if that's possible.

okalachev, 2017-09-11:
@bkueng, thanks for the answer!
So, changing any parameters in-flight officially should not be working? Or it causes some kind of 'undefined behaviour'?
How do I fly between different points in OFFBOARD mode with different velocities, then?

bkueng, 2017-09-11:
> So, changing any parameters in-flight officially should not be working? Or it causes some kind of 'undefined behaviour'?

It's definitely supported and we do it regularly. Just not multiple at (almost) the same time.

> How do I fly between different points in OFFBOARD mode with different velocities, then?

I have not used offboard, but the `SET_POSITION_TARGET_LOCAL_NED` message contains velocity fields.

okalachev, 2017-09-11:
>I have not used offboard, but the SET_POSITION_TARGET_LOCAL_NED message contains velocity fields.
Unfortunately, only position control **or** velocity control can be used, not both at the same time.

>It's definitely supported and we do it regularly. Just not multiple at (almost) the same time.
So changing one setting is OK, but simultaneously 2-3 or more is not OK?

bkueng, 2017-09-11:
> Unfortunately, only position control OR velocity control can be used, not both at the same time.

Isn't there another message to set the velocity then? @TSC21 might know.

> So changing one setting is OK, but simultaneously 2-3 or more is not OK?

Until now, yes. https://github.com/PX4/Firmware/pull/7950 fixes that. But setting parameters for your use-case is the wrong approach. It should be possible to do that via a mavlink message.

okalachev, 2017-09-11:
> Isn't there another message to set the velocity then? @TSC21 might know.
> setting parameters for your use-case is the wrong approach

I thought exactly the same earlier. In AUTO mode there is `MAV_CMD_DO_CHANGE_SPEED` message. But I couldn't find the way to change the maximum speed in position control in OFFBOARD mode. The only answer I got in PX4 Forum is [to use parameters](http://discuss.px4.io/t/proper-way-to-control-velocity-and-position-simultaneously-in-offboard/4119/4).

> Until now, yes
That seems like a critical issue!

Shpiler, 2017-09-11:
We applied [#7950](https://github.com/PX4/Firmware/pull/7950) for testing. It doesn't stop motors anymore, but looses OFFBOARD mode almost every time we change speed with interval less than 2s.

bkueng, 2017-09-12:
> It doesn't stop motors anymore, but looses OFFBOARD mode almost every time we change speed with interval less than 2s.

Thanks for testing. Like I wrote, it's not a good idea to constantly change parameters. The mavlink message spec of `SET_POSITION_TARGET_LOCAL_NED` does not exclude setting both velocity & position. But the position controller uses only one or the other ([here](https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L1216)). So we should consider extending support for that @Stifael.

Stifael, 2017-09-12:
The parameters `MPC_XY_VEL_MAX`, `MPC_Z_VEL_MAX_DN`, `MPC_Z_VEL_MAX_UP` are not meant to change during any flight. They are max-values which are respected by the controller. This means that if you set a velocity demand larger than MPC_XY_VEL_MAX, the demanded setpoint will be clipped at MPC_XY_VEL_MAX. It is kind of a safety implementation.

If you want to change velocity between waypoints, then you have to set the velocity demand accordingly (use `mavros/setpoint_velocity/cmd_vel`).

It is also correct that the vehicle can operate either in position control (which is a cascaded P controller with a PID velocity controller) or just the velocity controller. If you want to use position and velocity, one way to do it is by computing the velocity setpoint offboard by mimic the firmware position controller (which is just a simple P-controller). 

However, we are working on supporting both by adding velocity feedforward demands. That way position and velocity demands can be sent simultaneously.

Shpiler, 2017-09-12:
> If you want to use position and velocity, one way to do it is by computing the velocity setpoint offboard by mimic the firmware position controller (which is just a simple P-controller).

What rate the firmware position controller works at? It is easy to do it offboard, but we are limited by mavlink feedback rate. It some tasks like precision landing it can be a problem.

closed 2018-02-21 by PX4BuildBot