PX4/PX4-Autopilot #7941: MC: Estimator divergence?

tops4u, 2017-09-08:
**Bug Report**

Just had a short testflight for the LIDAR - I am aware that the battery was almost empty. I started over concrete and then put the copter over a corn field to check LIDAR Performance (I have seen before that it had some difficulties since it either toke ground or the full grown corn height as reference). Never the less the Copter started diverging in YAW about 30° without any Stick Input (this can be seen in the Log as the compass Values change) but the Yaw Angle and Angular Rate do not reflect this despite the Gyro also showing movement. Then the Aircraft started doing circles as it can be seen in the GPS Log and also the Estimated position. This also happened without any command from the RC. 

Can anybody explain this? Vibration Figures have been worse before but I have never seen this behaviour. 

https://logs.px4.io/plot_app?log=1bd0bb4b-020e-40e1-b0dd-10b8232d2b20

bkueng, 2017-09-11:
Which time are you refering to? Around 0:53? That was a (the only) 30 deg yaw change, but that was commanded via RC.
Looks more like an estimator problem. @priseborough 

Also make sure to disable ulog streaming in QGC. It's disabled by default and generally not needed if you use an SD card.

tops4u, 2017-09-11:
@bkueng no the YAW turn happened before it started circling around. I would guess around 0:42. You can see Gyro and Magnetometer changes without any RC Stick input. Just after switching to PosCTL.

I was not aware of this QGC Switch, I was thinking this stores the Log simply inQGC directly. But if this jams Telemetry I'll disable this.

bkueng, 2017-09-11:
I see. Due to the logging drop right before, it's difficult to say what's happening.

> I was not aware of this QGC Switch, I was thinking this stores the Log simply inQGC directly. But if this jams Telemetry I'll disable this.

Yes it saves it in QGC, but the telemetry does not nearly have enough bandwidth for that feature.

tops4u, 2017-09-11:
@bkueng Ok thanks.

- In your Opinion, shall I close this as it is not reproducible (not enough Information). 
- Can you tell me if this Problem (as well as #7949) does not occur in 1.6.5 or previous versions?

bkueng, 2017-09-11:
> In your Opinion, shall I close this as it is not reproducible (not enough Information).

Maybe someone else finds something. I'd leave it open for a bit but then close if it does not happen again.

> Can you tell me if this Problem (as well as #7949) does not occur in 1.6.5 or previous versions?

I cannot tell about this issue but I'm pretty sure that #7949 is a problem in previous versions too. #7949 is a problem for higher tilt angles (60+), your log only goes to around 30.

tops4u, 2017-09-26:
I'll close this for now.

closed 2017-09-26 by tops4u