PX4/PX4-Autopilot #7942: HRT time doesn't set the interrupt priority, will cause low priority interrupt response??

yangang123, 2017-09-09:
@file drv_hrt.c

```
static void
hrt_tim_init(void)
{
	/* claim our interrupt vector */
	irq_attach(HRT_TIMER_VECTOR, hrt_tim_isr);

	/* clock/power on our timer */
	modifyreg32(HRT_TIMER_POWER_REG, 0, HRT_TIMER_POWER_BIT);

	/* disable and configure the timer */
	rCR1 = 0;
	rCR2 = 0;
	rSMCR = 0;
	rDIER = DIER_HRT | DIER_PPM;
	rCCER = 0;		/* unlock CCMR* registers */
	rCCMR1 = CCMR1_PPM;
	rCCMR2 = CCMR2_PPM;
	rCCER = CCER_PPM;
	rDCR = 0;

	/* configure the timer to free-run at 1MHz */
	rPSC = (HRT_TIMER_CLOCK / 1000000) - 1;	/* this really only works for whole-MHz clocks */

	/* run the full span of the counter */
	rARR = 0xffff;

	/* set an initial capture a little ways off */
	rCCR_HRT = 1000;

	/* generate an update event; reloads the counter, all registers */
	rEGR = GTIM_EGR_UG;

	/* enable the timer */
	rCR1 = GTIM_CR1_CEN;

	/* enable interrupts */
	up_enable_irq(HRT_TIMER_VECTOR);
}
```
HRT time doesn't set the interrupt priority, will cause low priority interrupt response??
thank you !!!

mrpollo, 2017-10-18:
@davids5 please comment if we should keep this for 1.7

LorenzMeier, 2017-10-30:
Not critical for this release.

closed 2018-02-21 by PX4BuildBot

davids5, 2018-10-04:
For the record: Having a priority above normal could cause nesting. Nuttx does not support Nested interrupts, in fact enabling them will cause hard to debug hard faults. The common ISR is not reentrant. Tasks invoked off of the HRT ISR need to allow context switches. So the use of Nuttx's HIGH PRIORITY interrupts is not an options either.