PX4/PX4-Autopilot #7947: previous main state isn't changing

garfieldG, 2017-09-10:
Looking at state_machine_helper cpp file we can see that we call main_state_transition function from inside set_link_loss_nav_state function. When we do that we call it with a local variable main_state_prev.
In case that the main state has changed we save the current state as previous main state before changing the current state inside main_state_transition function. We save the previous main state in order to run a check on it inside the commander thread main loop. If we only change the local variable (instead the global variable declared in commander.cpp) we loose the information about the previous main state and it might effect in the commander thread main loop.
If we call the main_state_transition directly from the commander we do change the global variable.

dagar, 2017-09-14:
Yes, it's not ideal. The entire thing needs a cleanup, and the state machines should be much simpler. 

In this specific case I believe the actual usage of the previous main state is quite limited. https://github.com/PX4/Firmware/blob/master/src/modules/commander/commander.cpp#L2978

Unless you see an immediate fix should we try to account for this in the commander rewrite? https://github.com/PX4/Firmware/issues/7055

garfieldG, 2017-09-17:
@dagar 
Thanks for the response!

I think that it can be addressed while rewriting the commander.

closed 2017-09-18 by dagar