PX4/PX4-Autopilot #7955: Barometric Altimeter problem.

tubeme, 2017-09-12:
It is a corner case but still dangerous one.

By accident we put a pretty strong tape over the baro sensor on the board. It was tested on a plain Quad MC.

We created a mission with a take off alt 50 meters and the whole mission was at 50m height.

When we engaged the mission and the quad made a take off it never stopped climbing!!! It climbed for about 1 minute up to a point when we could not see the quad. The reading of the rel altitude stayed at 100-110m but the quad just climbed and climbed. It never started the mission to its 1 WP. 

We took manual control over the quad at this point and landed safely. Then we realized we closed the whole of the baro sensor so it could not have the chance to measure correctly the altitude.

We thought that it should fuse the GPS data as well in the altitude reading, but in our case it did not and the machine continued its climb in the clouds.

I will post the LOG in a couple of hours.

tubeme, 2017-09-14:
Here is the LOG:

https://logs.px4.io/plot_app?log=bcca2bcf-5d5b-48ac-b129-86aed4b6f0e0

dagar, 2017-09-21:
@priseborough anything we can do to check for this?

priseborough, 2017-09-24:
@tubeme Can you please check that you uploaded the correct log. The log you attached has only 19 seconds of data and the GPs data does not show the climb you described.

tubeme, 2017-09-25:
It is again this broken card. I have only this recorded from the flight...

tubeme, 2017-09-25:
It is the correct one just a portion of it...

closed 2018-02-21 by PX4BuildBot