PX4/PX4-Autopilot #7961: Full Parameter Load fails

tops4u, 2017-09-13:
**Bug Report**

I was trying Arducopter and saved all Parameters from 1.6.2dev in a file. After Re-Installing PX4 1.6.5dev (Master) I tried to reload all saved Parameters. This is not possible as the modification of certain parameters trigger a restart request. It seams that at this Point the Parameter Loading fails or stops. Redoing loading after a reset did not help in my situation.

Currently I ended up with a partially loaded Parameter List. For example Accelerometer 0 was correctly initialised and Temperature Coefficients are loaded, but Gyro Data was missing as well as the rest of all TC Data (Gyro, Baro, etc). 

I'm not sure if this is a QGC or PX4 Issue. QGC used is a daily build from 18.7.

dagar, 2017-09-13:
Do you know how far it got? We might ultimately need to handle it on the QGC side, but we should first figure out exactly what's happening.

tops4u, 2017-09-14:
@dagar I can retry, as I figured still a lot of Parameters missing in my current Config (for ex most of the ThermalCompensation Data, as well as some of the Basic Attitude Tuning Params). Would a Import an Re-Export be helpfull in order to check the differences? 
Should 1.6.2 Load into a 1.6.5dev be possible?

tops4u, 2017-09-15:
Here is what I did. I used an Pixracer and set it to default Values and upgraded to 1.6.5 Stable. 

After 1 Load and boot. Boot the following Parameters were exported. There was a Gyro Error.
After the 2nd Load and boot the following Parameters were exported and the Gyro Error gone. 
After the third, I did not check, but here are the three Files that have been exported afterwards.
Resulting Parameters
[Archiv.zip](https://github.com/PX4/Firmware/files/1307798/Archiv.zip)

Used Input:
[680Konfig.bak.params.zip](https://github.com/PX4/Firmware/files/1307809/680Konfig.bak.params.zip)

tops4u, 2017-09-17:
Forgot, here what QGC says:

[image]

dagar, 2017-09-17:
That actually makes more sense. On the fresh setup all those params aren't used/marked active until you calibrate. QGC doesn't "see" them and then doesn't load.

I need to look at the relevant QGC code, but I think it could be trying to set them regardless, and only then complaining if it fails.

dagar, 2017-09-17:
@DonLakeFlyer what do you think about the last comment? Could the QGC parameter manager try a PARAM_SET in this situation?

tops4u, 2017-09-17:
@dagar I would say that when loading Parameters from a File QGC should always try to write them and also disregard Params with a Reboot Flag until all are loaded.

DonLakeFlyer, 2017-09-18:
I'd have to look at the failure path. I think it would be ok. The reboot thing is red herring that has no affect other than popping the message. Can you put an issue in for QGC?

tops4u, 2017-09-19:
This is now logged as QGC Issue : https://github.com/mavlink/qgroundcontrol/issues/5672

closed 2017-09-19 by tops4u