PX4/PX4-Autopilot #7967: POSCTL problems on latest master build

Shpiler, 2017-09-14:
We've just flown latest master build since we need [#7950](https://github.com/PX4/Firmware/pull/7950) working on the drone as soon as possible, but met another issue. Drone doesn't fly normally in POSCTL mode - it just can't hold it's altitude. Like this: 
[image]

As I could see, altitude deviations had some correlation with wind gusts, but.. It certainly could fly normally in such conditions before with the same parameters (except what was added with firmware upgrade). Can we fix that problem or just use #7950 with stable release code?

Log: https://logs.px4.io/plot_app?log=57c6c42e-87e9-4b02-9000-b514d5aa3a7c

bkueng, 2017-09-15:
@Stifael can you look into that? I don't see anything wrong on the system level.

Stifael, 2017-09-15:
@Sphiler
can you upload another log from flights that worked as expected?
this log shows that the imu measurements in z is most likely clipping caused by vibration. I looked at the innovation, and the innovation for vel_z and pos_z are too large. In addition, both innovation have same sign, which again indicates bad imu measurements 
[image]

closed 2018-02-21 by PX4BuildBot