PX4/PX4-Autopilot #7991: Matrix library submoduled twice: in Firmware and in ecl

MaEtUgR, 2017-09-20:
During setup of this pr https://github.com/PX4/Firmware/pull/7908 I ran into unexpected problems.

After some analysis I found out the problem is the interpretation of the effect resulting from the matrix library being a submodule twice in the code base:
```
Firmware
├─ ecl
│  └─ matrix
└─ matrix
```

**Probably expected:** I and according to our discussion also @bkueng expected the calls to the matrix library inside the ecl library to use their instance and hence version of the submodule. And all the calls to the matrix library inside the Firmware but outside the ecl would use the submodule included in the Firmware directly.

**Reality:** I checked and **this is not the case**. Istead when compiling the Firmware **all** the calls go to the submodule included in the Firmware. In fact you can completely delete all the source content of ecl/matrix and it the compilation of the Firmware does not see any effect.

**Conclusion:** I think this is a problem for the code base structure. I know @priseborough wants to have the ecl self contained but this structure of submodules can easily lead to a lot of confusion.

FYI @LorenzMeier @julianoes @bkueng @dagar

dagar, 2017-09-20:
Even when not using EKF2 the Firmware requires ecl (data validator used by sensors, L1 controller, FW attitude controller). ECL requires the Matrix library. It seems to me Firmware should be getting Matrix through ECL.
@jgoppert would this still be an issue for you, or has Matrix stabilized sufficiently?

Additionally I think the standalone EKF2 build shouldn't be different than the way it's built for Firmware inclusion, but that can be a separate issue.

MaEtUgR, 2018-02-21:
This is still a problem which I cannot fix myself, are there no plans?

dagar, 2018-02-21:
We should make firm plans.

priseborough, 2018-02-22:
I'm happy to remove ecl/matrix provided we provide a documented way for developers wanting to incorporate ecl into their own projects to create a shared library. See current instructions in README.md

MaEtUgR, 2018-02-22:
That sounds like a plan.
About making ecl self contained, I never looked into that myself but @Stifael told me it's in the current state not that easy. He did some tests here I think: https://github.com/PX4/ecl/pull/355

MaEtUgR, 2018-04-30:
Hey @priseborough, is there any progress on this? We just had a new case of confusion resulting from the "double include".

Do you know who are the main clients of the ecl library? We could make very clear documentation how to use ecl by itself or what I would suggest a repo for self contained one-click usage of the latest ecl stable version that includes all necessary dependencies.

priseborough, 2018-04-30:
The paparazzi project was integrating it, but I do not know the current status. Also https://inertialsense.com was using it, but I have not heard from them in over 12 months.

dagar, 2018-04-30:
We should be able to drop matrix from the ecl project now and still have a standalone ecl build the pulls the matrix dependency from github if needed.

MaEtUgR, 2018-05-07:
@dagar I see you worked on this 🎉 
The double trouble module is gone with https://github.com/PX4/ecl/commit/e5952fadafc2de2ad5db841fd53ef7e9d85b5742#diff-8903239df476d7401cf9e76af0252622L1 in https://github.com/PX4/ecl/pull/432 and https://github.com/PX4/Firmware/pull/9406.
I hope @priseborough is content and was able to resolve his build system problems reported in the ecl pr. Otherwise we need to follow up on that.
But the original issue for the Firmware is resolved and hence I'm closing.

closed 2018-05-07 by MaEtUgR