PX4/PX4-Autopilot #7992: mocap/pose and local_position/pose not same

b-rohit, 2017-09-20:
I have connected rpi to pixhawk on telem2 port and publishing mocap data on `/mocap/pose` topic. The position value for ` z` is not same for `mocap/pose` and `local_position/pose`.
for mocap /pose>>
```
---
header: 
  seq: 1526
  stamp: 
    secs: 1505918137
    nsecs: 129957914
  frame_id: fcu
pose: 
  position: 
    x: 5.30004
    y: 1.2797
    **z: 1.96025**
  orientation: 
    x: 0.901538723658
    y: 0.425992599711
    z: -0.0661677012611
    w: 0.037149294027
---
```
local_position/pose
```
header: 
  seq: 1962
  stamp: 
    secs: 1505918137
    nsecs:  67410240
  frame_id: fcu
pose: 
  position: 
    x: 5.30003929138
    y: 1.27970004082
    **z: -0.049612943083**
  orientation: 
    x: 0.00923220180577
    y: -0.00385768606231
    z: -0.888187765534
    w: -0.459371736031
---

```
2. on roslauch mavros window , I keep getting following messages> 
```
[ INFO] [1505918071.621063490]: FCU: [lpe] mocap position init:  1.28,  5.30, -1.96 m 
[ INFO] [1505918071.671072088]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1505918071.720345636]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1505918071.770876366]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1505918071.822297508]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1505918071.870953039]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1505918071.922953865]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1505918071.971344190]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1505918072.023350910]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1505918072.070452714]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1505918072.120236745]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1505918072.172328734]: FCU: [lpe] reinit P (0, 0) negative

```

TobiMiller, 2017-10-09:
you need to set your height usage from barometer to mocap/vision ;)

b-rohit, 2017-10-10:
@TobiMiller Can I achieve that by changing value of the parameter LPE_FUSION ?

TobiMiller, 2017-10-10:
oh. you use lpe. try EKF2 and use the parameter EKF2_HGT_MODE

b-rohit, 2017-10-11:
@TobiMiller I am aware that `mocap` is only tested with `LPE` .

b-rohit, 2017-10-18:
I flew the drone 2 times today with mocap and it is very unstable. find logs here [logs1](https://logs.px4.io/plot_app?log=b4013090-7578-4315-be91-52d86187efd2) [logs2](https://logs.px4.io/plot_app?log=5ed74a8a-e39c-4fe9-9ba6-0ff9c66cdb91).  I keep getting following logs on roslaunch console

```
`[ INFO] [1508319837.645686588]: FCU: [lpe] mocap position init:  0.92,  0.37, -0.88 m 
[ INFO] [1508319837.697792949]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1508319837.746900766]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1508319837.796136994]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1508319837.844418005]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1508319837.896508119]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1508319837.946234558]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1508319837.997116121]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1508319838.046806681]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1508319838.100463979]: FCU: [lpe] reinit P (0, 0) negative
[ INFO] [1508319838.144437062]: FCU: [lpe] reinit P (0, 0) negative
`
```

TobiMiller, 2017-10-20:
go with ekf2 ;) i use it with no problems. the problems i discovered are fixed and was partly my failure

closed 2018-02-21 by PX4BuildBot

Abelteklu, 2019-10-31:
I'm having the same issue. Both in EKF2 and LPE. Has anyone solved this?