PX4/PX4-Autopilot #7998: PX4 1.6.5: No Disarm after mission complete.

tubeme, 2017-09-21:
I could not provide LOGs because the SD card broke so the files got corrupted but there are two problems in a consecutive mission flights of a plain QUAD multicopter frame.

1. After mission complete the quad did not disarm. We made sure the throttle lever is at 0 (we've had similar issue not disarming because of the Throttle position at the moment of touch down.). Waited for about 2 minutes, then switched to manual and disarmed.

2. On the second flight once the quad took off and gained the 50m takeoff (and mission) altitude the quad made 360 turn and then went towards the first WP. (We've seen this in VTOL in a previous 1.6 version once). SO this is the second time we see such a behavior. 

We will try to reproduce the flights very soon in order to have some LOGS.

dagar, 2017-09-21:
I'm now following the second problem. Is this a mission with a takeoff waypoint?

tubeme, 2017-09-21:
Yes both were missions (MC quad) with a takeoff point.

closed 2018-02-21 by PX4BuildBot