PX4/PX4-Autopilot #7999: master_new_archs_nuttx_upgrade build issues

davids5, 2017-09-21:
@dagar master_new_archs_nuttx_upgrade

- [ ] build issues VN
- [ ] V not working
- [ ] Hardlink edit not working

dagar, 2017-09-21:
Can you expand on hardlink edit not working?

davids5, 2017-09-21:
@dagar updated

dagar, 2017-09-25:
Hardlink copy was accidentally disabled (likely from testing OSX).

Touching defconfig and triggering a rebuild is working for me (with and without hardlink edit). Are you talking about through menuconfig?

davids5, 2017-09-25:
@dagar - no it was a disable of a I2C I was testing.

dagar, 2017-09-26:
Verbose mode readded. You can use V=1 or VERBOSE=1.

davids5, 2017-09-29:
@dagar - still some defconfig issues 

defconfig changes not in build

make clean
make px4fmu-v3_default

vim `# CONFIG_NSH_DISABLE_FREE not used`  => `CONFIG_NSH_DISABLE_FREE=y`

make px4fmu-v3_default

```
cat build/px4fmu-v3_default/NuttX/nuttx/.config | grep FREE
CONFIG_NSH_DISABLE_FREE=y
```
```
nsh> ?
help usage:  help [-v] [<cmd>]

  [           date        free        mount       rmdir       true
  ?           df          help        mv          set         umount
  break       echo        kill        mw          sh          unset
  cat         exec        ls          ps          sleep       usleep
  cd          exit        mkdir       pwd         test
  cp          false       mkfatfs     rm          time
```

dagar, 2017-10-02:
It's rebuilding for me.

	 % make px4fmu-v3_default   
	ninja: no work to do.

	 % vim nuttx-configs/px4fmu-v3/nsh/defconfig

	 % make px4fmu-v3_default                     
	 [868/868] Creating nuttx_px4fmu-v3_default.px4

	% grep -nr CONFIG_NSH_DISABLE_FREE build/px4fmu-v3_default 
	build/px4fmu-v3_default/NuttX/nuttx/.config:1555:CONFIG_NSH_DISABLE_FREE=y

davids5, 2017-10-02:
@dagar but when you run that version `free` should be gone and it is not.

dagar, 2017-10-02:
Oh okay, so it's rebuilding everything, but for each NuttX lib it's simply make in each directory. NuttX itself doesn't recognize that dependency.

dagar, 2017-12-18:
I believe all of these issues are resolved.

closed 2017-12-18 by dagar