PX4/PX4-Autopilot #8004: The local_position/pose topic's refresh rate is too low

wwwefwehfrgjnrek, 2017-09-24:
Hey, We are using a VICON system + Pixhawk on our uav. We have Configured the Firmware, and are able to get the mocap data on the `local_position/pose` topic. However, when I look at the publishing frequency of the topic's data ,it is found to continue to decrease even lower than 1HZ.Meanwhile, the situation of data loss occurs.I am very confused.Then I commented out the code which is subscribe to the image topics and the function that handles it.The topic 's publishing frequency is 10Hz and the data will not be lost.I do not know why this happens，how to solve it. Thank you

dagar, 2017-09-24:
Where are you seeing the rates? You likely need to increase it here. https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L2042

This can be done by choosing a different mavlink mode, asking for a higher rate via mavlink, or the command line. `mavlink stream -d /dev/ttyS1 -s LOCAL_POSITION_NED -r 100`

wwwefwehfrgjnrek, 2017-09-25:
@dagar Thanks for your help. I use the command  `rostopic hz /mavros/local_position/pose` to view the frequency of data published by mavros.The frequency of data published is  continue to decrease even lower than 1HZ.If i asking for a higher rate via mavlink,it will affect other data  published?

dagar, 2017-09-25:
That would completely depend on your setup, although you should definitely be able to stream much faster than that unless something is very wrong.

dagar, 2017-09-25:
I'll close this for now. If you have further questions please ask on http://discuss.px4.io/ or https://px4.slack.com.

closed 2017-09-25 by dagar