PX4/PX4-Autopilot #8010: Multi-instance Subscription

jlecoeur, 2017-09-25:
**Feature Request**

It would be nice to have an abstraction to easily subscribe to multiple instances of an uORB topic.

---
A good use case for this abstraction are position estimators that now subscribe to multiple instances of the topic `distance_sensor`. Right now, boiler plate code is duplicated in EKF2, LPE and INAV to do the following:
- loop over instances to check which one was updated
- check if the distance sensor fits some criteria. Namely, they check that the distance is within range, that the type of sensor is either SONAR or LIDAR, and that the sensor is downward facing.
- once an instance fits all criteria, then either:
    * (a) it is selected as THE distance sensor and is used for all following updates the filter.
    * (b) it is used once to update the filter, and all instances are checked again for following estimator updates 

Here are the corresponding code portions for ekf2, lpe and inav estimators. Some info is also in PR #7984 :
- [EKF2](https://github.com/PX4/Firmware/blob/16825e6460e5839ee5267ba1e825c8e3cf3f1b2f/src/modules/ekf2/ekf2_main.cpp#L560-L575). This case (a), it selects the first distance sensor that is downward facing. Then only uses this instance.
- [LPE](https://github.com/PX4/Firmware/blob/16825e6460e5839ee5267ba1e825c8e3cf3f1b2f/src/modules/local_position_estimator/BlockLocalPositionEstimator.cpp#L227-L252). This case (a) 2 times, it selects one distance sensor that is downward facing and is a SONAR, and a second instance that is downward looking and is a LIDAR. Then uses only these two instances.
- [INAV](https://github.com/PX4/Firmware/blob/16825e6460e5839ee5267ba1e825c8e3cf3f1b2f/src/modules/position_estimator_inav/position_estimator_inav_main.cpp#L543-L562). This is case (b), at each loop, the first updated instance that is a downward looking sensor is used.
---
Here is the proposal:
- Write an abstraction `SubscriptionMulti<TOPIC_s, N>` that works just like `Subscription<TOPIC_s>` (see [here](https://github.com/PX4/Firmware/blob/master/src/modules/uORB/Subscription.hpp#L130-L144)) but subscribes to N instances, and implements internally the boiler plate linked above. 
- It is configured in different modes to return data from any valid instance (like INAV), or from the one that was valid first (like LPE and EKF2). 
- It takes as optional argument a function pointer to filter messages. the function is in the form `bool is_valid(const TOPIC_s& msg);`. In the case of `distance_sensor`, the function returns true only if the orientation is correct, the distance is within range, and the type of sensor is correct (i.e. either SONAR or LIDAR)

Here is what is would look like from user's perspective:
```cpp
// class header
[...]
// Subscription
const int MAX_DISTANCE_SENSORS_INSTANCES = 4;
SubscriptionMulti<distance_sensor_s, MAX_DISTANCE_SENSORS_INSTANCES> _sub_distance_sensor;

// valid range sensor
static bool distance_sensor_is_valid(const distance_sensor_s& rng)
{
    if (rng.min_distance <= rng.current_distance &&
        rng.max_distance <= rng.current_distance &&
        rng..orientation == distance_sensor_s::ROTATION_DOWNWARD_FACING &&
        rng.type == distance_sensor_s::MAV_DISTANCE_SENSOR_ULTRASOUND) {
         return true;
     } else {
         return false;
     }
}
[...]
```
```cpp
// class constructor
[...]
MyClass():
   _some_field_1(),
   _sub_distance_sensor(ORB_ID(distance_sensor),                // topic id
                        &MyClass::distance_sensor_is_valid,     // pointer to function to check validity of data
                        Subscription::FIRST_VALID),             // only use the instance that was valid first, other option is Subscription::ANY
  _some_other_field()
[...]
```

```cpp
// usage of SubcriptionMulti
if (_sub_distance_sensor.updated()) {
    _sub_distance_sensor.update();
    
    // Check validity of new measurement (to include directly in updated()?)
    if (_sub_distance_sensor.is_valid()) {
        // get data
        distance_sensor_s& rng = _sub_distance_sensor.get();
        // update filter
        update_with_rng(rng);
    }
}
```
The method `is_valid` can be used in single-instance cases to filter invalid data.

---
Do you think that would be useful?  Feedback is welcome.

TSC21, 2017-09-25:
I like the idea. But wouldn't make more sense to check for the sensor ID instead of the instances? I know that currently the sensor ID is not properly implemented, but we could start from there.
We could build the ID with a code that merges the different data on its fields - something like the I2C address + type + direction - this was already addressed in https://github.com/PX4/Firmware/pull/7710.

jlecoeur, 2017-09-25:
While I see the point of having unique sensor ID, I do not see how it applies here.
The goal here is to make it easy to discover sensors that meet some criteria. By using unique sensor ID, you tie the estimator to a specific model of sensor and kill modularity.
However if filtering by sensor ID is really what you need, it can be done in the `distance_sensor_is_valid()` method.

dagar, 2017-09-25:
Good idea! What I'm not sure of is how general it really needs to be. Maybe a one off DistanceSensor class that extends Subscription and does this is sufficient?

I was thinking of doing something similar for the MC rate controller. https://github.com/PX4/Firmware/blob/master/src/modules/mc_att_control/mc_att_control_main.cpp#L997
It subscribes to all gyros directly, but only grabs data from the primary.

I suppose yet another option would be the sensors module subscribing to all distance sensors, and only republishing the chosen.

jlecoeur, 2017-09-25:
Inheriting Subscription might be a bit messy as there will be the uORB handle inherited from Subscription + an array of N-1 added ones. 
In fact a composition of N Subscription may be cleaner. I feel that writing a SubscriptionMulti class or a DistanceSensor class will take approx the same time (modulo a one-liner to specialize SubscriptionMulti for distance_sensor_s).
If you use it for gyros as well then the abstraction is preferable.

> module subscribing to all distance sensors, and only republishing the chosen.

All estimators do not select range finders with the same criteria, so that does not work. For gyros however, why not republishing the selected-and-corrected-gyro on a separate topic?

dagar, 2017-09-25:
Yes you're right composition is cleaner. I was mainly thinking the selection of each sensor is sufficiently different that it might as well be a special one off class.

In the gyro/rate controller case it's done that way to keep latency to an absolute minimum.

jlecoeur, 2017-09-25:
> In the gyro/rate controller case it's done that way to keep latency to an absolute minimum.

ok got it!

> that it might as well be a special one off class.

Maybe I just feel too nervous when I see code that rely on nullptr duplicated with small variations several times :-)

The true question is how many sensors would benefit from multi topics? I can see distance_sensor, gyro/acc/mag, gps, airspeed ... others?

dagar, 2018-01-22:
I still think this is a good idea, but I'm wondering if we should do it manually case by case? Create a multi subscription handler for the distance sensors, and another for the gyro case, and as we go see if something more general becomes apparent.

A generic multi-subscription with voter (like the sensors module does for gyro, accel, mag, baro) might also be helpful.

closed 2019-02-04 by stale[bot]