PX4/PX4-Autopilot #8013: Helicopter Mixer Problem.

tubeme, 2017-09-26:
Hello guys,

I'm trying to setup a swashplate helicopter with the mixer provided:

H: 3
T:      0   3000   6000   8000  10000
P:    500   1500   2500   3500   4500
# Swash plate servos:
S:      0  10000  10000      0  -8000   8000
S:    140  13054  10000      0  -8000   8000
S:    220  13054  10000      0  -8000   8000

The swashplate servor work together so the S: part looks OK.

But I have to touch a little the throttle servo and the swashplate vs throttle curve.

I cant seem to get the scaling of the T: and P:

Any ideas how the mixer scaling works?

tubeme, 2017-09-27:
After all it is straight forward procedure with the mixer:

T: for the throttle servo scaling

P: for the common pitch.

I was able to make the necessary changes to the swash plate movement with:

P: -3500 -1500 0 2000 4500

Which are the 5 points on the scale where you want it to be.

So it accepted negative values in the P: mixer and we got use of it.

The problem is the T: mixer for the throttle servo accepted only positive values in the scale.

so the 

T: 0 3000 6000 8000 10000

means the throttle servo goes from 1500us to 2000us. And in my case i need the servo to go from 1300us to 1900us.

This measn I should use

T: -2000 -500 1500 4000 9500

But with negative values the mixer just freezes the whole PWM bus and there is no servo movement if I use the "-" values in the T: mixer.

I have no other means to control the throw of the servos except the mixer. So I'm stuck... and most likely other people with gas helicopter.

dagar, 2017-09-27:
You might consider creating a PR for these changes, or as a new airframe. PX4 is barely used for traditional helicopters and likely needs work.

tubeme, 2017-09-27:
There is no need for new frame. Just a quick change of the mixer. I don't know who wrote the mixer. These changes to the mixer are applicable for 100% of the helicopters with servo throttle. If its Gas or Glow engine it will need this change, otherwise could not be flown with this mixer... I never created PR...

dagar, 2017-09-27:
Sounds fine to me if you think the current mixer isn't right.

For things like you can skip all the git setup and do it straight through the web interface.

Fork PX4/Firmware, find the file you want to edit, click the edit icon (pencil) in the top right hand corner. Then submit the pull request.

[image]

tubeme, 2017-09-27:
I edit the mixer TXT file without a problem. But then when I use negative values in the T: line the whole PX4 PWM line got dead.  So probably using negative values in the T: line makes the whole mixer to stop working.

I dont know which file to edit for this... If I knew I could make PR. Probably even editing it myself if I understand the code.

tubeme, 2017-09-27:
@bartslinger Any ideas??

bartslinger, 2017-09-28:
How did you connect the servo wires? What hardware are you using?

The throttle 'servo' needs to be connected to port 1.

tubeme, 2017-09-28:
The servo is connected to ch1.

The thing is that:

H: 3
T: -2000 -500 1500 4000 9500
P: -3500 -1500 0 2000 4500

makes the whole PX4 mixer to halt.

while

H: 3
T: T: 0 3000 6000 8000 10000
P: -3500 -1500 0 2000 4500

works perfectly well but half the way.

the T: scaling from 0 - 10000 is actually from 1500us to 2000us.

I'm missing the 1000us to 1500us travel.

I tried

H: 3
T: T: 0 5000 10000 15000 20000
P: -3500 -1500 0 2000 4500

But what it does is just get to 2000us at the middle of the throttle stick. So it cuts on my travel.

Any sense of what I wrote?

bartslinger, 2017-09-30:
I don't have the equipment at the moment to test it myself, and I struggle to find the scaling from mixer to pwm. For example, the multicopter mixer provides outputs in the range 0..1, but I cannot find where that scales to the full PWM range.

tubeme, 2017-10-01:
Look how the Airplane throttle mixer looks like:

M: 1
O:       10000  10000       0  -10000  10000
S: 0 3       0  20000  -10000  -10000  10000

And it works exactly like

M: 1
O:       10000   10000      0  -10000  10000
S: 0 0   10000   10000      0  -10000  10000

in terms of servo travel.

In the THrottle mixer there is an offset -10000 and then the travel is from 0..20000 ie from 1000us to 2000us.

bartslinger, 2017-10-01:
It's not related to that. Anyway I think I recognize the problem. It is still weird, because this setup was working for me (over a year ago). The only possible explanation I can think of, is that the ESC I used was adapting itself to 1500-2000 us range. I can try to fix it, but I won't be able to test it myself.

tubeme, 2017-10-01:
I can test it right away but if you can make me a binary for FMUv2. I have no way to compile.

LorenzMeier, 2017-10-07:
Try with latest in QGC.

closed 2017-10-07 by LorenzMeier