PX4/PX4-Autopilot #8016: Servo doesn't move

micheleportioli, 2017-09-26:
Hi everyone, 
I'm pretty new in px4 so you have to bear me if I have some basic doubts. I've problem to check if my PixHawk and my servo work properly. So I tried to run the servo from the terminal with "hwtest" or typing "pwm tests -c 1" (my servo is in MAIN OUT 1). No responses at all.
Should I upload the mixer even if I've just an output? If yes how can I do it?

Thank you very much in advance for your answers

dagar, 2017-09-26:
Did you configure an airframe?

micheleportioli, 2017-09-26:
Hi, no I didn't configure anythings. Thats exactly the point. Is there any possibility to control a servo without configuring an airframe? I tried to understand the rover_steering_control example and if I start the module over the console no output is generated. How is this example built up, respectively where are the actuators control output feeded to the mixer? (Is there a mixer/init file or how can I run it properly? Thanks for your help.

bresch, 2017-09-27:
You have to specify the PWM value (in microseconds) you want to set with the argument `-p`. e.g. `pwm test -c 1 -p 1600`. However, I'm not sure if this command still works.
In order to make it working, you need to load a mixer by selecting an airframe. You can select it in QGC or make a custom one following this [documentation](https://dev.px4.io/en/airframes/adding_a_new_frame.html).
A useful command to see the pwm channels values is `pwm info`.

bresch, 2017-09-29:
@micheleportioli I close this issue as it should work. Feel free to reopen it if `pwm test` is broken.

closed 2017-09-29 by bresch