PX4/PX4-Autopilot #8017: Run more instances of the EKF

CarlOlsson, 2017-09-27:
What's the status of running more than one instance of the EKF and then pick the output from the healthiest one? Has it been considered and dropped or could it be something to implement?

My current concern lies in detecting bad airspeed measurements. E.g. if the estimator is rejecting the GPS measurements it could be both due to bad GPS data or bad airspeed data that have screwed up the wind estimate. In this situation it is currently hard to automatically detect the problem, however if we could run two EKFs in parallel we could do things like:
One EKF is fusing airspeed measurements and data from GPS 1 (we have two)
The second EKF is not fusing airspeed measurements but is using synthetic sideslip fusion to estimate wind and gets data from the second GPS

dagar, 2017-09-28:
I've actually been considering when thinking about possible driver architecture changes. The current sensors pipeline is in the way of using differents accels, gyros, mags, baro in each EKF, but it could be changed.

How would you want to handle the configuration of each? First there's the configuration of which sensors for which EKF, then there's the per EKF parameters. It looks like nearly all parameters are passed to the EKF backend, but a few are used by the actual PX4 module. We should probably create some mechanism for generating sets of parameters with an index.

Finally how would you decide which to publish for the main topics (vehicle_attitude and vehicle_local_position)?

Antiheavy, 2017-10-03:
Seems like this would be useful with dual airspeed sensors as well.

closed 2018-02-21 by PX4BuildBot