PX4/PX4-Autopilot #8021: Documentation of offboard mode

Mike-Bal, 2017-09-28:
First of all, thanks for providing this nice OS autopilot software.

Now my question is, if there is any place, where the offboard interface - as supported by the pixhawk autopilot - is documented? By that I mean something along the lines of:

- If I want to do X, I have to send mavlink message M with parameters set to P?
- If I send mavlink message M with parameter set P, pixhawk will do X?
- Flag F in mavlink message M is ignored by the Pixhawk / only the followig combinations are possible ...
- Message M is supported but has the following problems: ...

So far, I have found dozens of bits and pieces of information scattered around different web-pages and forum threads. Some informationmay be outdated, some information is contradictory and most information is only concerned with a very specific scenario or problem (best I found so far are the examples in https://github.com/mavlink/c_uart_interface_example).   
Of course, the source code itself can be seen as a ground truth, but -  aside from the problem of understanding a unfamiliar code base - as this seems to be a moving target it is not always clear,if a specific behavior is desired or will be change with the next patch.

======
FYI: My sepcific problem at the moment is (in case someone could help me with that):  

- What messages (with which flags) do I have to send in order to move a multicopter with a velocity of `{vx,vy,vz, vyaw}` realtive to local ned coordinates?

hamishwillee, 2017-10-09:
@Mike-Bal  I'm currently working on [this doc](https://github.com/hamishwillee/px4_user_guide/blob/af0381dcdea989b6679200fe7a1b7c70468b82e5/en/flight_modes/offboard.md) (as part of https://github.com/PX4/px4_user_guide/pull/65 ).

This covers most of you want, though not in quite the format you have asked it.
1. Only two messages are supported: 
   * Control vehicle position or velocity (SET_POSITION_TARGET_LOCAL_NED).
   * Control vehicle attitude/orientation (SET_ATTITUDE_TARGET)
2. These are supported in copter, but not fixed wing (the whole mode doesn't work in fixed wing properly)
3. There is some message formatting in the DroneCore code base. It is C++ and I could understand it, so you should also be able to.

The SET_POSITION_TARGET_LOCAL_NED message allows you to control velocity, position and acceleration components (although in practice accel is apparently not properly supported on our vehicles).

Other than that, I would suggest you ask the question on Slack, perhaps tagging in MAVLink category.

Mike-Bal, 2017-10-10:
@hamishwillee: Thanks for the link. That PR looks pretty nice. 

One thing I'd like to see added is a description of the valid combinations in the different flag parameteres like `type_mask field` and `coordinate_frame`. I think I remember for example that some time back you only could specify all velocities together or none, but not just e.g. velocity in x direction. Also, are all cooridnate systems supported? 

> 3. There is some message formatting in the DroneCore code base. It is C++ and I could understand it, so you should also be able to.

Thank you for the suggestion. Could you provide a link to the specific file? 

In any case, while this might be enough to answer my questions, I hope this is not the long term state of affairs:
1) Even if most developers won't interact directly with the mavlink protocol, there should be a specification/documentation of the pixhawk interface somewhere (if only to provide a spec against which mavros and DroneCore can develop) and that interface is mavlink. 
2) An examplary usage of an interface is quite different from an actual interface spec, as it only shows you some things that are possible, but not, what else is possible and in particular, what is not possible. 

Please don't read this as a demand from my side to do anything (after all, it is not like I'm paying for anything here) just as user feedback that the project maintainers or other contributors might or might not incorporate.

hamishwillee, 2017-10-10:
@Mike-Bal 
1. FYI Updated [doc here](https://github.com/hamishwillee/px4_user_guide/blob/14a1c2e81f459eac93574cf29c6479c8404878bb/en/flight_modes/offboard.md). I plan to merge this following minor clarifications from domain experts. 
2. I agree that it would be good to document clearly what are the valid combinations of  `type_mask` and `coordinate_frame`. Once the current updates are in, feel free to [raise an issue against the new offboard doc](https://github.com/PX4/px4_user_guide/issues/new). The DroneCore implementation is not yet complete, but once it is I will be writing some test/example code for DroneCore that will make it easier to answer these questions and to track changes in behaviour. 
3. The cpp file [is here](https://github.com/dronecore/DroneCore/blob/develop/plugins/offboard/offboard_impl.cpp#L152)

> Please don't read this as a demand from my side to do anything (after all, it is not like I'm paying for anything here) just as user feedback that the project maintainers or other contributors might or might not incorporate.

Feedback is good :-) If you find out more as part of your experimentation please share with the community!

hamishwillee, 2017-10-12:
The offboard modes are live now here: https://docs.px4.io/en/flight_modes/offboard.html

The last addition was information about the supported frames.

I am closing this. @Mike-Bal If there are still specific outstanding issues with the doc then please raise an issue in the PX4 user guide repo (just click the bug link at top of the linked page). Note that there probably are still a few outstanding questions about edge cases, but we won't be able to answer those without testing (to come). But still worth adding an issue so we don't forget!

r0gelion, 2017-10-12:
@hamishwillee ,

Developer Resources->Offboard Control from Linux (PX4 Devguide) link is broken

hamishwillee, 2017-10-13:
Thanks very much @r0gelion - fixed by https://github.com/PX4/px4_user_guide/commit/c433bcebb2b28dcb2247bd265ec4eeca618b4952

hamishwillee, 2017-10-13:
Actually I can't close this, do not have rights.

closed 2017-10-13 by dagar

Mike-Bal, 2017-10-13:
@hamishwillee: Thank you very much I'll report back if anything important happens. One last question that came up: Do you know if there is a lower bound on the speed the autopilot will accept (e.g. all values smaller than 0.1m/s will be mapped to zero?)

hamishwillee, 2017-10-15:
@Mike-Bal No idea - a good question for the [discussion boards](http://discuss.px4.io/)