PX4/PX4-Autopilot #8025: PX4 version on Pixhawk

Sandra-M4M, 2017-09-29:
Hi everybody,

I'm having trouble identifying the release of PX4 I have flashed on my Pixhawk. I've followed instructions here: https://github.com/PX4/Firmware/issues/5854 but I'm still unable to identify the release.

I've opened de Mavlink Inspector and found the AUTOPILOT_VERSION segment, when I open it I'm able to read the flight_sw_version, but does not make sense to me the number 1.71717e+07 number written there, shouldn't it be 1.4.2 or something like that? Any help is greatly appreciated. I'm attaching a printscreen from the widget.

[image]

dagar, 2017-10-11:
Hi @Sandra-M4M,

Check the setup tab in QGC for a version number in a human readable form.
Alternatively you can check through the mavlink console (or serial console) run `ver all`.

dagar, 2017-10-11:
If you have further questions like this please ask us on slack (http://slack.px4.io/) or discuss (http://discuss.px4.io/).

closed 2017-10-11 by dagar