PX4/PX4-Autopilot #8027: Navigator size -> restructuring?

ChristophTobler, 2017-09-29:
Can we rethink the way the navigator works? Currently all the subclasses like Land, RTL, etc. inherit from MissionBlock (see e.g. https://github.com/YUNEEC/Firmware/blob/flow_takeoff_and_land/src/modules/navigator/land.h#L51). This dramatically increases the memory size of the Navigator (and more importantly unnecessarily).

I saw that @dagar already started on pushing things out of the MissionBlock [here](https://github.com/PX4/Firmware/pull/7932) :+1:. 

Is there maybe a way to get rid of the inheritance or shrink the MissionBlock even more?

The reason is that many platforms struggle with memory size and therefore also with new modules/features.

dagar, 2017-09-29:
We should test and get https://github.com/PX4/Firmware/pull/7932 in. After that there's still more we can get out of MissionBlock, but it's slightly more work. 

After that we could consider allocating the different navigation states more intelligently. https://github.com/PX4/Firmware/blob/master/src/modules/navigator/navigator.h#L285

For example DataLinkLoss and RCLoss are special outback challenge settings and hardly anyone uses them. GpsFailure is only used by FW and FollowTarget is only used by MC.

Stifael, 2017-09-29:
Would it not be possible that mission_block only contains a data structure similar to mission_item which is then inherited by all subclasses (RTL, LAND ...). All current mission_block related methods could be pushed to another class which contains the current mission_block methods.

dagar, 2017-09-29:
Similarly I've thought about not having MissionBlock as an interface at all. It's really only used by the current thing running to build up a mission_item, publish a position setpoint, and then check acceptance. Navigator only needs one of these, not 10.

We'd have to carefully check each navigation mode to make sure nothing is assumed to be conserved when leaving one state state and reentering later.

LorenzMeier, 2017-09-29:
I think we need a call to discuss this. Trying to turn Mission states into passive objects is increasing code complexity greatly because you loose the ability to specialize inside them.

Overall we should eliminate obvious wastes of RAM and flash, but make sure to not get too clever limiting our ability to innovate.

dagar, 2017-09-29:
Yes we should discuss it before changing anything. I wouldn't do it if it increased complexity or even lost any potential flexibility, but so far in the cases I've checked there's no need to keep that state when inactive.

dagar, 2017-09-29:
For reference

	INFO  [navigator] sizeof(_mission): 536
	INFO  [navigator] sizeof(_loiter): 376
	INFO  [navigator] sizeof(_rtl): 424
	INFO  [navigator] sizeof(_dataLinkLoss): 504
	INFO  [navigator] sizeof(_engineFailure): 344
	INFO  [navigator] sizeof(_gpsFailure): 424
	INFO  [navigator] sizeof(_rcLoss): 360
	INFO  [navigator] sizeof(_takeoff): 360
	INFO  [navigator] sizeof(_land): 344
	INFO  [navigator] sizeof(_follow_target): 936
	INFO  [navigator] 
	INFO  [navigator] sizeof(MissionBlock): 344
	INFO  [navigator] sizeof(Navigator): 6784

	nsh> free
		     total       used       free    largest
	Mem:        232240     207040      25200      20208
	nsh> navigator stop
	INFO  [navigator] exiting
	nsh> free
		     total       used       free    largest
	Mem:        232240     196000      36240      30672
	nsh>

Stifael, 2017-10-03:
about flexibility: as far as I understand the code, what actually defines a mission are the mission_items. For every mission, there can only be one mission_item active. In fact, each mission block follows that structure already: a mission block (such as rtl) only contains one mission_item, which gets refilled with new information once the current mission_item completes. This means that we could also just use one mission_item overall, which agrees with @dagar's comment:
> Similarly I've thought about not having MissionBlock as an interface at all.

closed 2018-02-21 by PX4BuildBot