PX4/PX4-Autopilot #8034: make: *** [px4fmu-v2_default] Error 2

bsowlin, 2017-09-30:
I reinstalled Ubuntu 16.04LTS on my notebook, so I have to reinstalled PX4 Toolchain but error happen when I tried 'make px4fmu-v2_default' (I've already 'make distclean') I have no idea what to do? it's shown like this

-- Build Type: MinSizeRel
-- PX4 VERSION: v1.6.5-1384-g0d5480e
-- CONFIG: nuttx-px4fmu-v2-default
-- The ASM compiler identification is GNU
-- Found assembler: /home/bsowlin/gcc-arm-none-eabi-5_4-2016q2/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2  
-- C compiler: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
-- Building and including px4io-v2
-- Configuring 
-- Version: 1.0.6
-- To change the version modify the file configure.ac
-- micro-CDR without check endianness
-- NuttX: px4fmu-v2 nsh cortex-m4
-- ROMFS: px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bsowlin/src/Firmware/build/px4fmu-v2_default
Scanning dependencies of target ver_gen
Scanning dependencies of target uorb_headers
Scanning dependencies of target lib__micro-CDR
[  1%] Generating git hash header
[  1%] Generating uORB topic headers
[  1%] Building C object src/lib/micro-CDR/src/c/CMakeFiles/lib__micro-CDR.dir/microCdr.c.obj
[  1%] Built target ver_gen
Scanning dependencies of target copy_px4io_bin
[  1%] Building px4io-v2
make[4]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
Scanning dependencies of target nuttx_configure
[  1%] Linking C static library liblib__micro-CDR.a
[  1%] Built target lib__micro-CDR
Scanning dependencies of target git_nuttx
[  1%] git submodule platforms/nuttx/NuttX/nuttx
[  1%] Copying NuttX/apps to /home/bsowlin/src/Firmware/build/px4fmu-v2_default/NuttX
[  1%] Built target git_nuttx
Scanning dependencies of target git_ecl
[  1%] git submodule src/lib/ecl
[  1%] Built target git_ecl
Scanning dependencies of target mixer_gen
[  1%] Generating mixer_multirotor.generated.h
[  1%] Built target mixer_gen
Scanning dependencies of target airframes_xml
[  1%] Creating airframes.xml
[  1%] Built target airframes_xml
Scanning dependencies of target parameters_xml
[  1%] Generating parameters.xml
[  1%] Copying NuttX/nuttx to /home/bsowlin/src/Firmware/build/px4fmu-v2_default/NuttX
-- Build Type: MinSizeRel
[  1%] Copying NuttX config px4fmu-v2 and configuring
-- PX4 VERSION: v1.6.5-1384-g0d5480e
-- CONFIG: nuttx-px4io-v2-default
-- The ASM compiler identification is GNU
-- Found assembler: /home/bsowlin/gcc-arm-none-eabi-5_4-2016q2/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
[  1%] Built target nuttx_configure
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2  
-- C compiler: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]
-- Configuring 
-- Version: 1.0.6
-- To change the version modify the file configure.ac
-- micro-CDR without check endianness
-- NuttX: px4io-v2 nsh cortex-m3
-- Configuring done
Scanning dependencies of target nuttx_context
[  1%] Built target uorb_headers
-- Generating done
[... 99 lines cut ...]
[ 96%] Built target modules__systemlib__mixer
Scanning dependencies of target nuttx_px4io-v2_default.elf
make[7]: *** No rule to make target 'NuttX/apps/libapps.a', needed by 'nuttx_px4io-v2_default.elf'.  Stop.
make[7]: *** Waiting for unfinished jobs....
[ 96%] Building C object src/firmware/nuttx/CMakeFiles/nuttx_px4io-v2_default.elf.dir/__/__/platforms/empty.c.obj
CMakeFiles/Makefile2:2260: recipe for target 'src/firmware/nuttx/CMakeFiles/nuttx_px4io-v2_default.elf.dir/all' failed
make[6]: *** [src/firmware/nuttx/CMakeFiles/nuttx_px4io-v2_default.elf.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make[5]: *** [all] Error 2
Makefile:166: recipe for target 'px4io-v2_default' failed
make[4]: *** [px4io-v2_default] Error 2
src/drivers/px4io/CMakeFiles/copy_px4io_bin.dir/build.make:93: recipe for target '../px4io-v2_default/nuttx_px4io-v2_default.elf' failed
make[3]: *** [../px4io-v2_default/nuttx_px4io-v2_default.elf] Error 2
CMakeFiles/Makefile2:1945: recipe for target 'src/drivers/px4io/CMakeFiles/copy_px4io_bin.dir/all' failed
make[2]: *** [src/drivers/px4io/CMakeFiles/copy_px4io_bin.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[  2%] Built target nuttx_arch_build
[  2%] Built target nuttx_drivers_build
[  2%] Built target nuttx_sched_build
Makefile:160: recipe for target 'all' failed
make[1]: *** [all] Error 2
Makefile:166: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 2

Appreciate every answer. Thank you.

dagar, 2017-09-30:
Hi @bsowlin, I wonder if you've hit on a Makefile problem. All the CI systems and most developers are using ninja-build instead of cmake.

Can you try ninja? `sudo apt-get install ninja-build` Then `make distclean` and `make px4fmu-v2_default`.

I'll try to reproduce this with regular Makefiles and find a fix.

bsowlin, 2017-10-01:
@dagar Thank you so much man. It's work!!!! and surprisingly. It's so much faster than regular one I love it.

dagar, 2017-10-08:
Yes it will use ninja by default if found.

closed 2017-10-08 by dagar

Jai-GAY, 2018-03-21:
sorry, can't follow as i have the same error. is it that my system should only contain ninja-build only ?

ninja: error: '../../platforms/nuttx/NuttX/nuttx/.git', needed by 'NuttX/git_init_nuttx.stamp', missing and no known rule to make it

$ ninja --version
1.6.0
$ ninja-build --version
1.7.2

mbangtri, 2018-09-12:
I got the same problem....currently I am using PixhawkPSP_v3.0.3. I try to build firmware, and the following error appear...

-- PX4 VERSION: v1.8.0-1010-gbf87270ded
-- CONFIG: nuttx_px4fmu-v2_default
-- Build Type: MinSizeRel
CMake Deprecation Warning at /usr/share/cmake-3.10/Modules/CMakeForceCompiler.cmake:69 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:30 (cmake_force_c_compiler)
  /usr/share/cmake-3.10/Modules/CMakeDetermineSystem.cmake:94 (include)
  CMakeLists.txt:174 (project)

CMake Deprecation Warning at /usr/share/cmake-3.10/Modules/CMakeForceCompiler.cmake:83 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:36 (cmake_force_cxx_compiler)
  /usr/share/cmake-3.10/Modules/CMakeDetermineSystem.cmake:94 (include)
  CMakeLists.txt:174 (project)

CMake Deprecation Warning at /usr/share/cmake-3.10/Modules/CMakeForceCompiler.cmake:69 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:30 (cmake_force_c_compiler)
  build/nuttx_px4fmu-v2_default/CMakeFiles/3.10.2/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:174 (project)

CMake Deprecation Warning at /usr/share/cmake-3.10/Modules/CMakeForceCompiler.cmake:83 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:36 (cmake_force_cxx_compiler)
  build/nuttx_px4fmu-v2_default/CMakeFiles/3.10.2/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:174 (project)

-- The ASM compiler identification is GNU
-- Found assembler: /home/mbangtri/gcc-arm-none-eabi-5_4-2017q2/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.15")
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2
-- C compiler: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-583-g56b8bb0
-- Building and including px4io-v2
-- NuttX: px4fmu-v2 nsh cortex-m4
-- ROMFS: px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/c/PX4/Firmware/build/nuttx_px4fmu-v2_default
ninja: Entering directory `/mnt/c/PX4/Firmware/build/nuttx_px4fmu-v2_default'
[3/658] Building px4io-v2
-- PX4 VERSION: v1.8.0-1010-gbf87270ded
-- CONFIG: nuttx_px4io-v2_default
-- Build Type: MinSizeRel
CMake Deprecation Warning at /usr/share/cmake-3.10/Modules/CMakeForceCompiler.cmake:69 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:30 (cmake_force_c_compiler)
  /usr/share/cmake-3.10/Modules/CMakeDetermineSystem.cmake:94 (include)
  CMakeLists.txt:174 (project)

CMake Deprecation Warning at /usr/share/cmake-3.10/Modules/CMakeForceCompiler.cmake:83 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:36 (cmake_force_cxx_compiler)
  /usr/share/cmake-3.10/Modules/CMakeDetermineSystem.cmake:94 (include)
  CMakeLists.txt:174 (project)
[... 13 lines cut ...]
  cmake/toolchains/Toolchain-arm-none-eabi.cmake:36 (cmake_force_cxx_compiler)
  build/nuttx_px4io-v2_default/CMakeFiles/3.10.2/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:174 (project)

-- The ASM compiler identification is GNU
-- Found assembler: /home/mbangtri/gcc-arm-none-eabi-5_4-2017q2/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.15")
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2
-- C compiler: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors) 5.4.1 20160919 (release) [ARM/embedded-5-branch revision 240496]
-- PX4 ECL: Very lightweight Estimation & Control Library v0.9.0-583-g56b8bb0
-- NuttX: px4io-v2 nsh cortex-m3
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/c/PX4/Firmware/build/nuttx_px4io-v2_default
ninja: Entering directory `/mnt/c/PX4/Firmware/build/nuttx_px4io-v2_default'
[202/202] Creating /mnt/c/PX4/Firmware/build/nuttx_px4io-v2_default/px4io-v2_default.px4
[11/658] Generating parameters.xml
FAILED: parameters.xml
cd /mnt/c/PX4/Firmware/build/nuttx_px4fmu-v2_default/src/lib/parameters && /usr/bin/python /mnt/c/PX4/Firmware/src/lib/parameters/px_process_params.py --src-path /mnt/c/PX4/Firmware/src/lib/DriverFramework/framework /mnt/c/PX4/Firmware/src/lib/airspeed /mnt/c/PX4/Firmware/src/lib/battery /mnt/c/PX4/Firmware/src/lib/bezier  [... cut ...] /src/modules/mc_pos_control/mc_pos_control_tests --xml /mnt/c/PX4/Firmware/build/nuttx_px4fmu-v2_default/parameters.xml --inject-xml /mnt/c/PX4/Firmware/src/lib/parameters/parameters_injected.xml --overrides {} --board px4fmu-v2
Duplicate parameter definition: BAT_CAPACITY+
[14/658] Copying NuttX/nuttx to /mnt/c/PX4/Firmware/build/nuttx_px4fmu-v2_default/NuttX
ninja: build stopped: subcommand failed.
Makefile:154: recipe for target 'px4fmu-v2_default' failed
make: *** [px4fmu-v2_default] Error 1

byoruk, 2020-01-08:
> I got the same problem....currently I am using PixhawkPSP_v3.0.3. I try to build firmware, and the following error appear...
>
> -- PX4 VERSION: v1.8.0-1010-gbf87270ded
> -- CONFIG: [...]

Again that problem any answer for it ?
ninja: fatal: chdir to '/cygdrive/d/mypx4/Firmware/build/nuttx_px4fmu-v3_default' - No such file or directory
make: *** [Makefile:154: px4fmu-v3_default] Error 1

psp316r, 2020-05-22:
@algebratech i too have same Problem, did you get any solution by now ?