PX4/PX4-Autopilot #8035: **Feature Request**  param to enable auto transition to Mission after successful Takeoff

Antiheavy, 2017-10-01:
Right now if we want to auto takeoff and auto fly a Mission everything is handled inside the flight plan including auto takeoff parameters built into the first mission waypoint.  This is the case for fixed wing at least, I'm not sure about VTOL.

Another (and IMHO better) way to do this is to use the Takeoff modes/states and then automatically switch to the Mission once Takeoff is achieved.  There are many use cases where this is a a better approach.  This should be enabled/disabled by a parameter.

Another important piece to this is properly parameterizing the fixed wing takeoff pitch: 
 https://github.com/PX4/Firmware/issues/6787

dagar, 2017-10-05:
I'm likely going to squeeze this into https://github.com/PX4/Firmware/issues/6787

Then a param like COM_TAKEOFF_FIN for configuring what happens after AUTO_TAKEOFF acceptance. Default would be switch to LOITER, another option would be switch to MISSION (if available and valid).

Antiheavy, 2017-10-05:
related: https://github.com/PX4/Firmware/pull/8020

Antiheavy, 2017-10-05:
Another advantage of using "Takeoff" mode instead of takeoff waypoint in Mission mode is the airplane can back off from 100% throttle sooner instead of climbing all the way to mission altitude at 100% throttle.  Long climbs at 100% throttle, especially the takeoff climb, result in reduced airspeed/pitch control, can be inefficient, and can overtemp the motor/esc on some aircraft.

LorenzMeier, 2017-10-30:
@dagar I've moved this to v1.8 - please move back to 1.7 and send a PR if you disagree.

Antiheavy, 2017-10-30:
@dagar  - FWIW this is not urgently needed for our project so we have no issues delaying.

Antiheavy, 2017-11-17:
@dagar we should review to ensure mission feasibility properly gates both Takeoff and Mission modes is auto-transition from Takeoff to Mission is active.

closed 2018-02-21 by PX4BuildBot

Antiheavy, 2018-02-21:
I'm okay with this feature request getting closed.  I believe if https://github.com/PX4/Firmware/issues/6787 is implemented then both Takeoff mode and Mission takeoff modes will work slightly differently and will answer the mail on this.  There may not need to be a parameter to force auto transition, just use Mission takeoff in that case instead of Takeoff mode.