PX4/PX4-Autopilot #8053: RC Blips when adding a 2nd mavlink stream

r1b4z01d, 2017-10-04:
While hovering 4 feet off the ground the drone rolled 30 degrees left and would not recover with RC controls. After analysing the logs and live mavlink data the drone was acting to what it thought was an RC command. As we debugged we observed jumps in raw rc controls. We tried a separate RC receiver as well as a separate transmitter with the same issues. Analysing the raw inputs via QGroundControl we walked through our config file and removed line by line params. We determined the that multiple mavlink streams were the cause of the issue. 

QGroundControl Raw RC Blips: https://i.imgur.com/BaFnAE6.png

Hardware:  EAGLE
Software Version:  f160743d
OS Version: Linux, v3.4.0
Log from flight: https://logs.px4.io/plot_app?log=f38f504d-2888-422a-a86a-52c60f00b449

dagar, 2017-10-04:
Is this with spektrum RC (https://github.com/PX4/Firmware/tree/master/src/drivers/spektrum_rc)?

If so we probably need to bump the priority - https://github.com/PX4/Firmware/blob/master/src/drivers/spektrum_rc/spektrum_rc.cpp#L223
https://github.com/PX4/Firmware/blob/master/src/platforms/px4_tasks.h#L146

r1b4z01d, 2017-10-04:
That is correct we are using the Spektrum RC.

ndepal, 2017-10-05:
I have seen similar blips, which have taken the vehicle out of mission mode without me knowing it. Also with a Spektrum DSMX.

dagar, 2017-10-05:
@ndepal would that have been a pixracer in your case?

ndepal, 2017-10-05:
No it was a PixFalcon

dagar, 2017-10-08:
It's the same DSM library, but different "drivers". In the Snapdragon case I'd still try bumping the priority as I described.

@ndepal if you have a log take a look at input_rc when it happened.

ndepal, 2017-10-09:
Here is a log of it happening: http://logs.uaventure.com/view/ZBWRhaatFHLTcVdYkvrTzS

This is on pretty old firmware (it's on a frozen demo setup). You can see some spikes in multiple RC channels which cause the flight mode to change out of mission mode.

closed 2018-02-21 by PX4BuildBot