PX4/PX4-Autopilot #8056: MC: PWM goes into Saturation in PosCTL

tops4u, 2017-10-05:
Today I had my first crash on my Tarot 680. I have mounted new ESC (Hobbywing XRotor 35A Mini with BLHeli on DampedModeLight) for TMotor 4010/475kV Motors with 13x4.4 Carbon Props.

I started in Stabilized Mode and wanted to do a fast flight over a field. Then I switched to PosCtl Mode as I have been doing many times before. In this moment I could see at least one Rotor come to a full stop. The Vehicle started tumbling and started falling. In order to prevent more damage I flipped the Kill - Switch before it was out of sight.

In the Log File I can see that the PWM was going to Min PWM Value (1000us) respectively to Max PWM. This will cause Motors to full stop with the new ESC. I have observed that my current BLHeli Configuration may have trouble restarting Motors when they are stopped and restarted to a non minimal Speed. So I assume the ESC was unable to restart some of the Motors after they have been stopped with a non minimal PWM Speed setting. Despite this Problem IMHO the FC should in no case go to 1000us while in flight state.

I was assuming that MPC_THR_MIN / MPC_THR_MAX should avoid this behavior. They are on default values. 12% / 90% respectively. So min PWM should be limited to Values between 1120 and 1900 in all Auto Thrust Controll Modes... Anybody got an explanation for this?

Damage: Landing Gear, Landing Legs, Rear Right Arm (Motor, ESC and Props needs checking), 2 Video TX at least...

Logfile: https://logs.px4.io/plot_app?log=182a1539-78b6-40dc-88e7-9540fcd9f19a

This is a copy of http://discuss.px4.io/t/crash-after-switching-to-posctl/4588/1 since I consider this either a configuration fault or major bug.

LorenzMeier, 2017-10-05:
The fix for that is not the min and max throttle but PWM_MIN and PWM_MAX which should be set such that motors never stop.

tops4u, 2017-10-05:
@LorenzMeier I'm rather confused now. I have PWMMin and PWMMax @Default Values. I perform a ESC Calibration. My assumption was that there is nothing more to do as there is MPC_THR_MIN and MPC_MANTHR_MIN both should in my Opinion (and understanding by the comment for those) prevent that PWM ever reach MIN or MAX... Please enlighten me.

tops4u, 2017-10-16:
@LorenzMeier I try to understand this concept with PWM_MIN in combination with MPC_THR_MIN and MPC_MANTHR_MIN. 
As to my understanding by reading the Parameters commentaries several Times IMHO the following Workflow should be ok, when setting up a new Vehicle:
- Set PWM_MIN / PWM_MAX to defaults (as Comment suggests this is Industry Standard). 
- Perform ESC Calibration (as far as I have seen ESC Calibration is using PWM_MIN / PWM_MAX for Calibration). 
- Check that *THR_MIN are not set to 0 in order to not switch the motors off completely in Flight. 

Currently I don't understand (and I did not find any other info on this in the Docs) why I have to manually augment PWM_MIN so that the ESC just do not stop completely when there are the two above mentioned Parameters for MPC which should in my opinion do exactly the same thing. I could find references for the above Params in POS_CTL Module and the Landing Detector.

tops4u, 2017-11-04:
Will create a PR for Documentation Change regarding PWM Settings in order to prevent this situation.

closed 2017-11-04 by tops4u