PX4/PX4-Autopilot #8059: Standard way to disable internal magnetometer?

EliaTarasov, 2017-10-05:
I didn't find standard parameter to disable internal magnetometer. Does it exist or the only solution is to disable it in lsm303d.cpp?

dagar, 2017-10-05:
There isn't currently a good way to do this (discussed a bit here: https://github.com/PX4/Firmware/pull/8039).

For now you could hack the init script (https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rc.sensors#L142) and not start the driver.

EliaTarasov, 2017-10-06:
Thanks dagar, i've already modified the code as short-term hack.

closed 2017-10-06 by EliaTarasov

IeiuniumLux, 2018-06-28:
@ dagar, what is the parameter in v1.8.0 to disable the internal magnetometer in the Pixracer?

hamishwillee, 2018-06-28:
Possible unset https://docs.px4.io/en/advanced_config/parameter_reference.html#CAL_MAG1_EN ? @dagar It would actually be good to clarify this.

limhyon, 2019-01-05:
Hi,
I have not tried this, but, is it possible to set CAL_MAG1_EN before calibration?
For example, put it on a RC script. At the moment of parameter initialization, it will not be applied.