PX4/PX4-Autopilot #8066: Mission randomly rejected

sanderux, 2017-10-05:
Log: https://review.px4.io/plot_app?log=6f9fa451-d3a6-42dd-b1c7-87261310b139
Situation:
Vehicle sat outside with good gps and telemetry lock
Armed via RC
Switched from Manual to Mission (jumping over ALTCTL)
Warning issued: REJECT AUTO MISSION

Then disarm and arm again and it works.
This happens very often but not consistently.

jinchengde, 2017-10-18:
I also meet this issue in 1.6.5 of Fixedwing

dagar, 2017-10-19:
If you're able to reproduce getting a log with with mission_result, offboard_mission, and vehicle_status_flags added to logger would be helpful.

dagar, 2017-10-19:
The home_position as well.

jinchengde, 2017-10-21:
is that caused by  excessive vibration levels ? It was happened in gas engine but we haven't see in electrical

LorenzMeier, 2017-10-30:
@sanderux Does this persist with master?

sanderux, 2017-10-31:
Yes,
However, my switch does pass through ALTCTL quickly before mission is activated.
@dagar throught it might be related.

sanderux, 2017-10-31:
https://logs.px4.io/plot_app?log=5746a789-a331-4bd3-90c8-24d99c5a6520

dagar, 2017-11-02:
The mission didn't update until after the attempted switch to mission mode.

[image]

sanderux, 2017-11-03:
Must be a bug, there was no mission update at that time

mrpollo, 2017-11-08:
@dagar should we move this to 1.8?

jinchengde, 2017-11-11:
the same issue with log
https://logs.px4.io/plot_app?log=da1efbe4-9940-4b3e-a287-6b0aa8fb8af9

dagar, 2017-11-12:
Thanks for the logs, please keep posting and include any context. 

Side note - @jinchengde you have estimator issues https://logs.px4.io/plot_app?log=da1efbe4-9940-4b3e-a287-6b0aa8fb8af9#Nav-Estimator-Watchdog

jinchengde, 2017-11-13:
@dagar 
This is based on the firmware version of 1.6.5, and we always get this issue. then we rollback to 1.6.4, the problem is disappear, so I supposed it maybe caused by EKF ?

and could you tell me where you see the estimator issues in log.Thanks!

dagar, 2017-11-13:
@jinchengde I'm not sure, but there have been many robustness improvements for FW in EKF2 over the last few months. Give master a try (nearly v1.7.0) and open an issue or ping me if you're still seeing anything in the estimator watchdog.

sanderux, 2017-11-17:
Latest insights is that it has something to do with the fact that my switch (from manual to mission) travels over another flightmode (ALTCTL). If is switch slowly from manual - altctl - mission the problem is gone.

dagar, 2017-11-17:
Ok, I think it's just the timing of the mission_result publication. Need to think about it overall, but we need to handle that differently.

closed 2018-02-21 by PX4BuildBot