PX4/PX4-Autopilot #8068: SITL with gazebo and px4

SimonVleugels, 2017-10-05:
Hi,

I'm trying to simulate a drone controlled via a ros node but after a full day of trying, searching and reading i can't get it to work. 
I think I have installed everything
- mavros
- gazebo
- PX4 Frimware

When i try no_sim=1 make posix_sitl_default gazebo it will give me `INFO  [simulator] Waiting for initial data on UDP port 14560. Please start the flight simulator to proceed..`, so i guess that works fine too. So now i'm at the part of making my .launch file but there seems to be a problem. 
The launch file I have:
`<launch>

    <arg name="headless" default="false"/>
    <arg name="gui" default="true"/>
    <arg name="enable_logging" default="false"/>
    <arg name="enable_ground_truth" default="true"/>
    <arg name="enable_mavlink_interface" default="true"/>
    <arg name="ns" default="iris"/>
    <arg name="log_file" default="$(arg ns)"/>
    <arg name="mavlink_bridge_url" default="udp://127.0.0.1:14551@14555" />

    <include file="/home/simon/Documents/masterproef/masterproef_ws/Firmware/launch/gazebo_iris_empty_world.launch">
        <arg name="headless" value="$(arg headless)"/>
        <arg name="gui" value="$(arg gui)"/>
        <arg name="enable_logging" value="$(arg enable_logging)" />
        <arg name="enable_ground_truth" value="$(arg enable_ground_truth)" />
        <arg name="enable_mavlink_interface" value="$(arg enable_mavlink_interface)" />
        <arg name="log_file" value="$(arg log_file)"/>
    </include>
    <include file="/home/simon/Documents/masterproef/masterproef_ws/mavros/mavros/launch/mavlink_bridge.launch">
    <arg name="mavlink_bridge_url" value="$(arg mavlink_bridge_url)" />
    <arg name="ns" value="$(arg ns)"/>
    </include>
    <node pkg="ros_gazebo" type="gazebo_node" name="gazebo_node" output="screen"/>

</launch>`

I used the guide from: [https://dev.px4.io/en/simulation/ros_interface.html?q=](https://dev.px4.io/en/simulation/ros_interface.html?q=), [https://404warehouse.net/2016/07/11/px4-software-in-the-loopsitl-simulation-on-gazebo/](https://404warehouse.net/2016/07/11/px4-software-in-the-loopsitl-simulation-on-gazebo/) and [https://github.com/PX4/Firmware/issues/3368](https://github.com/PX4/Firmware/issues/3368) 

Can someone help me with this?

closed 2017-10-06 by SimonVleugels

YuYangguang, 2017-10-13:
Did you solve this issue? I also met this problem

dronecrewkari, 2018-01-18:
Above problem occur the incorrect listening port configuration.

https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/ekf2/iris#L49

Change the above line into 

```
simulator start -s -u 14560
```

This will make the listening port to become 14560 UDP port of PX4 firmware from Simulator.

hamishwillee, 2018-01-29:
Normally you should not need to change the config as above because the sim is automatically started looking at 14560. I *think* that doesn't happen in this case because we're building with  `no_sim=1`. 
Instructions for using simulators are here: https://dev.px4.io/en/simulation/