PX4/PX4-Autopilot #8069: Dangerous failsafe mode transitions

MarcGyongyosi, 2017-10-05:
Hi, 

We recently discovered some fairly dangerous behavior while experimenting on a robot running v1.6.5 stable on a Pixhawk Mini. The result is the robot arming and throttling up (i.e. taking off uncontrolled) without any way to stop it.

Here's exactly how this happens (we've replicated it a few times now):
1) Provide external vision position input
2) Fly successfully in offboard mode
3) Shut down external vision input/companion computer -> px4 goes into failsafe
4) Turn off RC transmitter (without leaving offboard mode) -> after a few seconds, the PX4 arms!!! and shoots off! 

At this point there is no way to control the px4 since both the companion computer and RC transmitter are off and disconnected.

While it is obviously not good practice for the transmitter to be turned off while the px4 is still in offboard mode, I think especially indoors it is something that can happen easily and thus DANGEROUS. Generally, I think it is a bad idea for the system to arm when no external controller/commander is connected to it. 

Luckily, nobody got injured this time but it could have ended pretty badly...

Hope this can be helpful and/or prevent future accidents. 

-marc & ifm

dagar, 2017-10-05:
Was the vehicle landed? Do you have a log?

MarcGyongyosi, 2017-10-06:
As I mentioned on the slack channel as well, the vehicle was on the ground when it armed.

Here is a log from when we turned off the onboard computer just as we were about to take off. It correctly disarmed given that the computer turned off. However, once the RC was turned off as well, the motors armed again and the thing went flying off. 

https://logs.px4.io/plot_app?log=f6c47702-efdb-49ea-8bfe-34e2bf1590f4

dagar, 2017-10-06:
According to vehicle_status the system only had small blips of lost RC, however from input_rc we can see that RC was lost (disconnected).

[image]

Notice input_rc.input_source is jumping between 3 for Spektrum (https://github.com/PX4/Firmware/blob/master/msg/input_rc.msg#L4) and 6 for Mavlink.

The autopilot was likely receiving RC_CHANNELS_OVERRIDE (http://mavlink.org/messages/common#RC_CHANNELS_OVERRIDE), which prevented RC lost mode.
https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_receiver.cpp#L1564

dagar, 2017-10-06:
I propose dropping RC_CHANNELS_OVERRIDE support. There are better mechanisms for manual control over mavlink.

visrinivasan, 2017-10-11:
Any updates here @MarcGyongyosi ?

I came across a similar situation today.
I have added a OFFBOARD Control Switch on my Remote Controller.
I have changed the offboard_mode sample script a little bit such that I don't set the OFFBOARD mode in my script but instead ARM it in when set to OFFBOARD with my RC.

I have been constantly checking the state of my Drone and it is initally in POSCTL.
When I turn ON the OFFBOARD switch, it goes to OFFBOARD and then switches back to POSCTL when I turn off the OFFBOARD switch.

Then I disarmed the drone using the RC. 

However, surprisingly, the drone ARMed after roughly 5mins without me arming it through the RC or QGC or putting the drone in OFFBOARD mode.

The only thing being my OFFBOARD script was running in the background, but it cannot arm from the script because it waits for the drone to go in OFFBOARD mode.

Luckily, I didnt have my propellers ON. Any pointers on what could have gone wrong? OR How can it be fixed?

dagar, 2017-10-12:
@visrinivasan I can take a look if you have log.

visrinivasan, 2017-10-19:
Thanks @dagar , I found the issue in my case.
It was in the offboard script. I fixed that up and now it doesn't create the anomaly.

dagar, 2017-10-19:
Thanks for the update. Was it related to RC_CHANNELS_OVERRIDE?

visrinivasan, 2017-10-19:
No, it wasn't.
It was a small bug in the ARMing logic of my OFFBOARD script.

closed 2018-02-21 by PX4BuildBot