PX4/PX4-Autopilot #8070: Nuttx missing tag nuttx-7.21 in repo

davids5, 2017-10-06:
@dagar I just noticed that ver is displaying  7.21.0, so I think the tag from  nuttx-7.22 git dropped. 

OS: NuttX
OS version: Release 7.21.0 (118817023)

Would you re-assign it from the commit on BB or change the parser to use the highest  tag in the form 'upstream_7.22+'?

dagar, 2017-10-08:
What do we need the upstream_X.Y+ tags for? Can we do it automatically from the real nuttx tags? 

We're currently missing the nuttx-7.22 tag.
https://github.com/PX4-NuttX/nuttx/releases

davids5, 2017-10-09:
upstream_X.Y+  It represents the point in upstream we added are patches on and "tied" Firmware too.

davids5, 2017-10-09:
I do not think we can. Tags from upstream are on the release, but immediately there are commits, hence the +.

dagar, 2017-10-10:
I pushed the missing upstream_7.22+.

closed 2017-10-30 by LorenzMeier