PX4/PX4-Autopilot #8073: MAV_DO_CHANGE_SPEED not applied correctly

sanderux, 2017-10-06:
This:
https://github.com/PX4/Firmware/blob/b4755297ecf3d91b2aa0d6d08d78e35f6692177d/src/modules/fw_pos_control_l1/FixedwingPositionControl.cpp#L774-L788

makes so that a MAV_DO_CHANGE_SPEED command is only applied when changing waypoints.

dagar, 2017-10-06:
How does that make it only apply when changing? The mission_throttle is passed to TECS each iteration here. https://github.com/PX4/Firmware/blob/b4755297ecf3d91b2aa0d6d08d78e35f6692177d/src/modules/fw_pos_control_l1/FixedwingPositionControl.cpp#L807

sanderux, 2017-10-06:
it sets it hard: mission_throttle = _parameters.throttle_cruise; 
Then only checks if there was a possp throttle set
So updating mission_throttle doesn't do anything

https://github.com/PX4/Firmware/blob/3c18be387cc47c9875137745b171e7ccec901f2e/src/modules/navigator/navigator.h#L204

same for airspeed

sanderux, 2017-10-06:
Ps: that is called from here in response to a do_change_speed command: https://github.com/PX4/Firmware/blob/3c18be387cc47c9875137745b171e7ccec901f2e/src/modules/navigator/navigator_main.cpp#L506

sanderux, 2018-02-23:
This is still an issue

goodman110110110, 2018-09-02:
I use firmware1.4.4, and I don't want to use the other version, how can I use do_change_speed command to change the mission speed?

hamishwillee, 2018-09-02:
@goodman110110110 Please do not append to old and somewhat unrelated issues. 

You really should update to the newest version, but to answer your question, you need to check if the mission item is supported in the version you want to use. If not then you'll have to add support in the code yourself. This will effectively be a custom message. See https://dev.px4.io/en/middleware/mavlink.html

goodman110110110, 2018-09-07:
Thankyou @hamishwillee

closed 2019-02-12 by stale[bot]