PX4/PX4-Autopilot #8087: PX4_simple_app not in Builtin App List?

KochC, 2017-10-08:
Hey, 
I try to create an own application for my DroPix, so I started with the "Hello Sky" Tutorial. First my compiler wants a CMakeList.txt file which is not described in the tutorial, but I was able to create one on my own 😆

px4_simple_app.c
```
#include <px4_config.h>
#include <px4_tasks.h>
#include <px4_posix.h>
#include <unistd.h>
#include <stdio.h>
#include <poll.h>
#include <string.h>

#include <uORB/uORB.h>
#include <uORB/topics/sensor_combined.h>
#include <uORB/topics/vehicle_attitude.h>

__EXPORT int px4_simple_app_main(int argc, char *argv[]);

int px4_simple_app_main(int argc, char *argv[])
{
    PX4_INFO("Hello Sky!");
		
    return OK;
}
```

CMakeLists.txt
```
px4_add_module(
	MODULE examples__px4_simple_app
	MAIN px4_simple_app
	STACK_MAIN 2000
	SRCS
		px4_simple_app.c
	DEPENDS
		platforms__common
	)
```

nuttx_px4fmu-v2_default.cmake
```
# Tutorial code from
# https://px4.io/dev/px4_simple_app
examples/px4_simple_app
```

Compiler prints correctly compiled:
```
-- Building and including px4io-v2
-- Configuring 
-- Version: 1.0.6
-- To change the version modify the file configure.ac
-- micro-CDR without check endianness
-- NuttX: px4fmu-v2 nsh cortex-m4
-- ROMFS: px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /Firmware/build/px4fmu-v2_default
```

and the upload seems to work:

```
Firmware chris$ make px4fmu-v2_default upload
[0/1] uploading /Firmware/build/px4fmu-v2_default/nuttx_px4fmu-v2_default.px4
Loaded firmware for 9,0, size: 1031060 bytes, waiting for the bootloader...
If the board does not respond within 1-2 seconds, unplug and re-plug the USB connector.
Attempting reboot on /dev/tty.usbmodem1 with baudrate=57600...
If the board does not respond, unplug and re-plug the USB connector.
Found board 9,0 bootloader rev 4 on /dev/tty.usbmodem1
ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff ffffffff type: ÿÿÿÿ
idtype: =FF
vid: ffffffff
pid: ffffffff
coa: //////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////8=

sn: 002d00323334510b32363238

Erase  : [====================] 100.0%
Program: [====================] 100.0%
Verify : [====================] 100.0%
Rebooting.
```

but I do not have the px4_simple_app in my Builtin Apps now:
```
nsh> help
help usage:  help [-v] [<cmd>]

  [           date        free        mount       rmdir       true        
  ?           df          help        mv          set         umount      
  break       echo        kill        mw          sh          unset       
  cat         exec        ls          ps          sleep       usleep      
  cd          exit        mkdir       pwd         test        
  cp          false       mkfatfs     rm          time        

Builtin Apps:
  adc
  bl_update
  bst
  camera_feedback
  camera_trigger
  commander
  config
  cu
  dataman
  ekf2
  ets_airspeed
  fmu
  frsky_telemetry
  fw_att_control
  fw_pos_control_l1
  gnd_att_control
  gnd_pos_control
  gpio_led
  gps
  hardfault_log
  hmc5883
  l3gd20
  land_detector
  lis3mdl
  ll40ls
  load_mon
  logger
  lsm303d
  mavlink
  mc_att_control
  mc_pos_control
  mixer
  mpu6000
  mpu9250
  ms4525_airspeed
  ms5525_airspeed
  ms5611
  mtd
  navigator
  nshterm
  param
  perf
  pwm_input
  pwm
  pwm_out_sim
  px4flow
  px4io
  reboot
  rgbled
  sdp3x_airspeed
  send_event
  sensors
  sercon
  serdis
  sf0x
  teraranger
  tone_alarm
  top
  ulanding_radar
  uorb
  ver
  vmount
  vtol_att_control
nsh> 
```

Where is my fault? 😩🤔

dagar, 2017-10-08:
It's probably my fault, not yours. For now try `make distclean` then build and upload again.
There's a problem with the dependencies for the nuttx builtin apps generation.

KochC, 2017-10-08:
nope, still not in the Builtin Apps list 😣
The folder `px4fmu-v2_default/src/examples/px4_simple_app`  is correctly generated in the build directory. I also flashed via terminal and QGroundControl.

KochC, 2017-10-10:
its still not possible to add any example to the Builtin Apps list. Is the tutorial wrong or is this bug?

dagar, 2017-10-10:
Are you on the current firmware master?

This is working for me.

	  px4_daemon_app
	  px4flow
	  px4_mavlink_debug
	  px4_simple_app
	  reboot
	  rgbled
	  rover_steering_control
	  sd_bench
	  sdlog2
	  sdp3x_airspeed
	  segway
	  send_event
	  sensors
	  sercon
	  serdis
	  sf0x
	  sf0x_tests
	  sf1xx
	  snapdragon_rc_pwm
	  srf02
	  tap_esc
	  teraranger
	  test_ppm
	  tests
	  tone_alarm
	  top
	  uavcan
	  uorb
	  uorb_tests
	  ver
	  vmount
	nsh> px4_simple_app
	INFO  [px4_simple_app] Hello Sky!
	INFO  [px4_simple_app] Accelerometer:     0.3223         -0.5603        -10.0705
	INFO  [px4_simple_app] Accelerometer:     0.3002         -0.6057        -10.1572
	INFO  [px4_simple_app] Accelerometer:     0.3050         -0.5334        -10.1045
	INFO  [px4_simple_app] Accelerometer:     0.2662         -0.5301        -10.0652
	INFO  [px4_simple_app] Accelerometer:     0.3151         -0.5478        -10.1150
	INFO  [px4_simple_app] exiting
	nsh> ver all
	HW arch: PX4FMU_V4
	FW git-hash: 8a0a8e20e1ce1a764300c99d1a28fbe6db15cfae
	FW version: 4.97.8 80 (73468032)
	OS: NuttX
	OS version: Release 7.22.0 (118882559)
	OS git-hash: 85a88c073130c0f9dd83af0a67e61c445ab0b3fa
	Build datetime: Oct 10 2017 10:23:01
	Build uri: localhost
	Toolchain: GNU GCC, 4.9.3 20150529 (prerelease)
	MFGUID: 363539313535510800320046
	MCU: STM32F42x, rev. 3
	UID: 320046:35355108:36353931 
	nsh>

KochC, 2017-10-10:
Ok, I hope thats it. Works now. Thanks a lot.

closed 2017-10-10 by KochC