PX4/PX4-Autopilot #8088: Error while executing “make posix jmavsim”

sudakshin, 2017-10-08:
Whenever i try to run "make posix jmavsim" in ~/src/Firmware it exits with this error.

CMake Error: The source directory "/home/sudakshin/src/Firmware/build/posix_sitl_default/-Wno-deprecated" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
/bin/sh: 1: cd: can't cd to /home/sudakshin/src/Firmware/build/posix_sitl_default
make: *** [posix_sitl_default] Error 2

mje-nz, 2017-10-11:
I have this error since 01b3e6f (merge #7873), on Mint 17.1 (Ubuntu 14.04). I'm trying to build px4fmu-v2_default. It's fine in the docker container though (px4io/px4-dev-nuttx). @sudakshin: What OS are you using?

YuYangguang, 2017-10-12:
I also have this problem, but when I git reset to v1.6.5, everything goes fine. Is there anyone could tell us why?

sudakshin, 2017-10-12:
@mje-nz : I'm using Ubuntu 14.04 and ros indigo
@YuYangguang : Which command do you execute exactly?

mje-nz, 2017-10-12:
@YuYangguang: What OS are you using?

YuYangguang, 2017-10-13:
I am using Ubuntu14.04 . You can use this command in firmware folder:
git reset --hard f160743df40b9dcc9558f3df1c6eeaf56646a758

the command above will reset the firmware to v1.6.5

YuYangguang, 2017-10-14:
Update your cmake to 3.9, this problem will be solved. 
Here is the cmake 3.9:
https://cmake.org/download/ @sudakshin

sudakshin, 2017-10-15:
@YuYangguang : Did installation of cmake version 3.9 but now i'm getting this error related to `ninja`

```
sudakshin@sudakshin:~/src/Firmware$ make posix_sitl_default gazebo
[3/7] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   thread
--   signals
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
--   atomic
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "2.5.0") 
-- Boost version: 1.54.0
-- Looking for OGRE...
-- Checking for module 'OGRE'
--   Found OGRE, version 1.8.1
-- Found Ogre Byatis (1.8.1)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   chrono
--   date_time
--   atomic
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
-- catkin DISABLED
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread;-lpthread (found version "2.5.0") 
-- Checking for module 'protobuf'
--   Found protobuf, version 2.5.0
Gazebo version: 7.8
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   chrono
--   date_time
--   atomic
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sudakshin/src/Firmware/build/posix_sitl_default/build_gazebo
[4/7] Performing build step for 'sitl_gazebo'
FAILED: cd /home/sudakshin/src/Firmware/build/posix_sitl_default/build_gazebo && /usr/local/bin/cmake --build . && /usr/local/bin/cmake -E touch /home/sudakshin/src/Firmware/build/posix_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-build
[1/2] Building CXX object CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o
FAILED: /usr/lib/ccache/c++  -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"mavlink_sitl_gazebo\" -Dgazebo_geotagged_images_plugin_EXPORTS -I/opt/ros/indigo/include -I/home/sudakshin/src/Firmware/Tools/sitl_gazebo/include -I/home/sudakshin/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/include/gazebo-7 -I/usr/include/sdformat-4.3 -I/usr/include/ignition/math2 -I/usr/include/OGRE -I/usr/include/OGRE/Terrain -I/usr/include/OGRE/Paging -I/usr/include/gazebo-7/gazebo/msgs -I/usr/include/eigen3 -isystem /usr/local/include -isystem /usr/local/include/opencv -I. -I/usr/include/eigen3/eigen3 -I/usr/local/include/OGRE -I/usr/local/include/OGRE/Paging -I/home/sudakshin/src/Firmware/Tools/sitl_gazebo/../../mavlink/include -std=c++11 -std=c++11 -Wno-deprecated-declarations -fPIC -MD -MT CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o -MF CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o.d -o CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o -c /home/sudakshin/src/Firmware/Tools/sitl_gazebo/src/gazebo_geotagged_images_plugin.cpp
/home/sudakshin/src/Firmware/Tools/sitl_gazebo/src/gazebo_geotagged_images_plugin.cpp: In member function ‘void gazebo::GeotaggedImagesPlugin::handle_message(mavlink_message_t*)’:
/home/sudakshin/src/Firmware/Tools/sitl_gazebo/src/gazebo_geotagged_images_plugin.cpp:350:43: error: too few arguments to function ‘uint16_t mavlink_msg_command_ack_pack_chan(uint8_t, uint8_t, uint8_t, mavlink_message_t*, uint16_t, uint8_t, uint8_t, int32_t, uint8_t, uint8_t)’
                                        100);
                                           ^
In file included from /home/sudakshin/src/Firmware/Tools/sitl_gazebo/../../mavlink/include/mavlink/v2.0/common/common.h:1184:0,
                 from /home/sudakshin/src/Firmware/Tools/sitl_gazebo/../../mavlink/include/mavlink/v2.0/common/mavlink.h:32,
                 from /home/sudakshin/src/Firmware/Tools/sitl_gazebo/include/gazebo_geotagged_images_plugin.h:32,
                 from /home/sudakshin/src/Firmware/Tools/sitl_gazebo/src/gazebo_geotagged_images_plugin.cpp:18:
/home/sudakshin/src/Firmware/Tools/sitl_gazebo/../../mavlink/include/mavlink/v2.0/common/./mavlink_msg_command_ack.h:110:24: note: declared here
 static inline uint16_t mavlink_msg_command_ack_pack_chan(uint8_t system_id, uint8_t component_id, uint8_t chan,
                        ^
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
make: *** [posix_sitl_default] Error 1

YuYangguang, 2017-10-16:
Hello. This issue seems to be a bug in the Firmware of current version. It seems that the file :" Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp" could not find the corresponding mavlink cpps. 

Here is what I do:
1.copy the folder "v2.0" in "Firmware/mavlink" to "Firmware/Tools/sitl_gazebo/include"
2. open the file "gazebo_mavlink_interface.cpp" and add a line:
#include "v2.0/common/mavlink.h"
3. build the Firmware, everything should be OK.

good luck @sudakshin

sudakshin, 2017-10-18:
@YuYangguang : Done, still the same error,  actually I referred to this issue ( #7858 ) and did a `git reset --hard 58fb126` on  `/Firmware/Tools/sitl_gazebo` due to which I was getting this error.
before performing a `git reset` I was getting this error.
```
sudakshin@sudakshin:~/src/Firmware$ make posix_sitl_default gazebo
[3/7] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   thread
--   signals
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
--   atomic
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "2.5.0") 
-- Boost version: 1.54.0
-- Looking for OGRE...
-- Checking for module 'OGRE'
--   Found OGRE, version 1.8.1
-- Found Ogre Byatis (1.8.1)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   chrono
--   date_time
--   atomic
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
-- catkin DISABLED
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread;-lpthread (found version "2.5.0") 
-- Checking for module 'protobuf'
--   Found protobuf, version 2.5.0
Gazebo version: 7.8
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   system
--   thread
--   timer
--   chrono
--   date_time
--   atomic
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sudakshin/src/Firmware/build/posix_sitl_default/build_gazebo
[4/7] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[7/7] cd /home/sudakshin/src/Firmware.../src/Firmware/build/posix_sitl_default
/bin/sh: 1: /home/sudakshin/src/Firmware/Tools/sitl_run.sh: Permission denied
FAILED: cd /home/sudakshin/src/Firmware/build/posix_sitl_default/tmp && /home/sudakshin/src/Firmware/Tools/sitl_run.sh /home/sudakshin/src/Firmware/build/posix_sitl_default/px4 posix-configs/SITL/init/ekf2 none gazebo none /home/sudakshin/src/Firmware /home/sudakshin/src/Firmware/build/posix_sitl_default
ninja: build stopped: subcommand failed.
make: *** [posix_sitl_default] Error 1
```

MansyrevAY, 2018-01-17:
Have the same problem on Windows 7, tried hard reset and it didn't help. Moreover, "git clone" command clones empty "Firmware/mavlink/v2.0" folder. Do I need to clone it separatly?

hamishwillee, 2018-01-29:
@MansyrevAY 

> Do I need to clone it separatly?

No, after you clone firmware, all the submodules should automatically get pulled when you build. 

Windows isn't currently properly supported. Suggest you try in a VM - the [linux scripts](https://dev.px4.io/en/setup/dev_env_linux.html) should work out of the box on Ubuntu 16.04.

Jai-GAY, 2018-03-22:
i was having an issue previously when trying to do [make posix jmavsim](https://dev.px4.io/en/setup/building_px4.html) today after trying [make submodulesclean] (https://github.com/PX4/Firmware/issues/9124#issuecomment-374863764)  +
 make distclean and re-do make posix jmavsim on CentOS 7.x, i am able to reach here.

[537/596] Building CXX object src/modules/fw_pos_control_l1/CMakeFiles/modules__fw_pos_control_l1.dir/Landingslope.cpp.o
[538/596] Linking CXX static library src/modules/fw_pos_control_l1/libmodules__fw_pos_control_l1.a
[539/596] Building CXX object src/modules/gnd_att_control/CMakeFiles/modules__gnd_att_control.dir/GroundRoverAttitudeControl.cpp.o
FAILED: cd /home/user/px4/github/Firmware/build/posix_sitl_default/src/lib/mixer && /usr/bin/python /home/user/px4/github/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py --normalize -f /home/user/px4/github/Firmware/src/lib/mixer/geometries/dodeca_bottom_cox.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/dodeca_top_cox.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/hex_cox.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/hex_plus.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/hex_t.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/hex_x.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/octa_cox.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/octa_cox_wide.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/octa_plus.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/octa_x.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_deadcat.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_h.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_plus.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_s250aq.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_vtail.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_wide.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_x.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_x_pusher.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_y.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/tri_y.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/twin_engine.toml -o mixer_multirotor_normalized.generated.h
Traceback (most recent call last):
  File "/home/user/px4/github/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py", line 336, in <module>
    A, B = geometry_to_mix(geometry)
  File "/home/user/px4/github/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py", line 182, in geometry_to_mix
    At = geometry_to_thrust_matrix(geometry)
  File "/home/user/px4/github/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py", line 166, in geometry_to_thrust_matrix
    Ct=np.array([[rotor['Ct']] for rotor in geometry['rotors']])).T
  File "/home/user/px4/github/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py", line 155, in thrust_matrix
    ax = axis / np.linalg.norm(axis, axis=1)[:, np.newaxis]
TypeError: norm() got an unexpected keyword argument 'axis'
FAILED: cd /home/user/px4/github/Firmware/build/posix_sitl_default/src/lib/mixer && /usr/bin/python /home/user/px4/github/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py -f /home/user/px4/github/Firmware/src/lib/mixer/geometries/dodeca_bottom_cox.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/dodeca_top_cox.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/hex_cox.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/hex_plus.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/hex_t.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/hex_x.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/octa_cox.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/octa_cox_wide.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/octa_plus.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/octa_x.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_deadcat.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_h.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_plus.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_s250aq.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_vtail.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_wide.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_x.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_x_pusher.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/quad_y.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/tri_y.toml /home/user/px4/github/Firmware/src/lib/mixer/geometries/twin_engine.toml -o mixer_multirotor.generated.h
Traceback (most recent call last):
  File "/home/user/px4/github/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py", line 336, in <module>
    A, B = geometry_to_mix(geometry)
  File "/home/user/px4/github/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py", line 182, in geometry_to_mix
    At = geometry_to_thrust_matrix(geometry)
  File "/home/user/px4/github/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py", line 166, in geometry_to_thrust_matrix
    Ct=np.array([[rotor['Ct']] for rotor in geometry['rotors']])).T
  File "/home/user/px4/github/Firmware/src/lib/mixer/geometries/tools/px_generate_mixers.py", line 155, in thrust_matrix
    ax = axis / np.linalg.norm(axis, axis=1)[:, np.newaxis]
TypeError: norm() got an unexpected keyword argument 'axis'
ninja: build stopped: subcommand failed.
make: *** [posix_sitl_default] Error 1

closed 2018-08-06 by santiago3dr