PX4/PX4-Autopilot #8093: "make posix_sitl_default gazebo" error

leomatrix, 2017-10-09:
when I try to use gazebo for px4, there is a build error!  
plus, I have insall gazebo7 successfully...

/home/leo/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp: In member function ‘void gazebo::GazeboMavlinkInterface::handle_message(mavlink_message_t*)’:
/home/leo/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp:903:8: error: ‘MAVLINK_MSG_ID_HIL_ACTUATOR_CONTROLS’ was not declared in this scope
   case MAVLINK_MSG_ID_HIL_ACTUATOR_CONTROLS:
        ^
/home/leo/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp:904:5: error: ‘mavlink_hil_actuator_controls_t’ was not declared in this scope
     mavlink_hil_actuator_controls_t controls;
     ^
/home/leo/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp:905:52: error: ‘controls’ was not declared in this scope
     mavlink_msg_hil_actuator_controls_decode(msg, &controls);
                                                    ^
/home/leo/Firmware/Tools/sitl_gazebo/src/gazebo_mavlink_interface.cpp:905:60: error: ‘mavlink_msg_hil_actuator_controls_decode’ was not declared in this scope
     mavlink_msg_hil_actuator_controls_decode(msg, &controls);
                                                            ^
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
make: *** [posix_sitl_default] Error 1

dagar, 2017-10-17:
Which version of PX4 Firmware did you try? Can you make sure you're on the latest master, run `make distclean` and try again?

leomatrix, 2017-11-06:
I can make sure that I'm the lastest code base and run "make distclean" & "make clean"
Plus I can run "make posix_sitl_default jmavsim " successfully

hiro64, 2017-12-05:
@leomatrix Did you manage to solve it?

hamishwillee, 2017-12-05:
What OS are you on too? 

The dev guide toolchain instructions use scripts to set up everything for you for Ubuntu  - see [here](https://dev.px4.io/en/setup/dev_env_linux.html#development-toolchain). This includes Gazebo. I tried this recently and it worked fine building "end-to-end".

leomatrix, 2017-12-29:
http://discuss.px4.io/t/no-such-file-or-directory-uorb-topics-sensor-combined-h/1336  
the above discuss is the same problem with me

hamishwillee, 2018-01-02:
@leomatrix I think the error in that link occurs because they are trying to build the helloworld apps when they haven't built PX4. The result is that the uORB headers haven't been generated. You have the same symptoms but the reasons why they have not been built may be different. Is your gazebo environment set up using our scripts?

leomatrix, 2018-01-12:
after update to ubuntu16.04.3LTS and re-install the tool chain the problem was fixed

closed 2018-01-12 by leomatrix

alexcherpi, 2018-05-15:
Hi,

I cloned the PX4 Firmware v1.7.3.
I am able to launch the jmavsim simulator with the command "make posix_sitl_default jmavsim" and it works fine.
But when I run the command "make posix_sitl_default gazebo" I get the following error log:

[1/1] Re-running CMake...
-- Build Type: RelWithDebInfo
-- PX4 VERSION: v1.7.3
-- CONFIG: posix_sitl_default
-- Build Type: RelWithDebInfo
-- C compiler: cc (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609
-- C++ compiler: c++ (Ubuntu 5.4.0-6ubuntu1~16.04.9) 5.4.0 20160609
-- Configuring done
-- Generating done
-- Build files have been written to: /home/alexcherpi/Documents/GIT_PolyDrones/polydrones_px4/build/posix_sitl_default
[7/18] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   thread
--   signals
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
--   atomic
-- Boost version: 1.58.0
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so (Required is at least version "2.3.0") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.1.4
-- Checking for module 'uuid'
--   Found uuid, version 2.27.0
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 2.2.0
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- FreeImage.pc not found, we will search for FreeImage_INCLUDE_DIRS and FreeImage_LIBRARIES
-- Checking for module 'gts'
--   Found gts, version 0.7.6
-- Checking for module 'libswscale'
--   Found libswscale, version 3.1.101
-- Checking for module 'libavdevice >= 56.4.100'
--   Found libavdevice , version 56.4.100
-- Checking for module 'libavformat'
--   Found libavformat, version 56.40.101
-- Checking for module 'libavcodec'
--   Found libavcodec, version 56.60.100
-- Checking for module 'libavutil'
--   Found libavutil, version 54.31.100
-- Checking for module 'jsoncpp'
[... 174 lines cut ...]
   math::Quaternion C_W_I = T_W_I.rot;
   ^
/home/alexcherpi/Documents/GIT_PolyDrones/polydrones_px4/Tools/sitl_gazebo/src/gazebo_imu_plugin.cpp:251:22: error: ‘C_W_I’ was not declared in this scope
   orientation->set_x(C_W_I.x);
                      ^
/home/alexcherpi/Documents/GIT_PolyDrones/polydrones_px4/Tools/sitl_gazebo/src/gazebo_imu_plugin.cpp:274:3: error: ‘math’ has not been declared
   math::Vector3 acceleration_I = link_->GetRelativeLinearAccel() - C_W_I.RotateVectorReverse(gravity_W_);
   ^
/home/alexcherpi/Documents/GIT_PolyDrones/polydrones_px4/Tools/sitl_gazebo/src/gazebo_imu_plugin.cpp:277:3: error: ‘math’ has not been declared
   math::Vector3 angular_vel_I = link_->GetRelativeAngularVel();
   ^
/home/alexcherpi/Documents/GIT_PolyDrones/polydrones_px4/Tools/sitl_gazebo/src/gazebo_imu_plugin.cpp:279:41: error: ‘acceleration_I’ was not declared in this scope
   Eigen::Vector3d linear_acceleration_I(acceleration_I.x,
                                         ^
/home/alexcherpi/Documents/GIT_PolyDrones/polydrones_px4/Tools/sitl_gazebo/src/gazebo_imu_plugin.cpp:282:38: error: ‘angular_vel_I’ was not declared in this scope
   Eigen::Vector3d angular_velocity_I(angular_vel_I.x,
                                      ^
ninja: build stopped: subcommand failed.
ninja: build stopped: subcommand failed.
Makefile:147: recipe for target 'posix_sitl_default' failed
make: *** [posix_sitl_default] Error 1

Do you know what's wrong ?
If I run gazebo it's working fine on its own.
My gazebo version is Gazebo9 and I m working on ubuntu 16.04LTS