PX4/PX4-Autopilot #8100: review calibration error messages

dagar, 2017-10-09:
QGC hides calibration messages (containing "[cal]"), but this also includes important errors, like out of memory. We need to review all calibration code and remove the [cal] prefix from anything critical a user actually needs to see.

https://github.com/PX4/Firmware/blob/master/src/modules/commander/mag_calibration.cpp#L578

dagar, 2017-10-09:
https://github.com/PX4/Firmware/issues/8098

dagar, 2017-10-13:
TODO: if USB is connected (extra mavlink running) and calibration doesn't have sufficient memory then suggest disconnecting.

bryanmonti, 2017-10-14:
Disabling the TELEM2 port also helps free up some memory (~2k). I also removed the IIC LED, didn't measure the memory difference though. I noticed more failures while connected over the telem radios, had good success over USB. More investigation is needed there.

Maybe we can integrate a check that checks for where the external mag is located (ie, UAVCAN or IIC) and disable IO that does not contain a mag. This is more of a temporary patch than a solution.

bryanmonti, 2017-10-17:
Saved a bit more memory by going with static node ID allocation vs the dynamic ID allocation server for UAVCAN. May put this in as the recommended route if people run into memory issues when using CAN.

dagar, 2018-01-22:
Done.

closed 2018-01-22 by dagar