PX4/PX4-Autopilot #8105: Sudden Jerk after takeoff causes crash (Kneejerk) - PX4 1.6.5px

mwingerson, 2017-10-10:
I have been fighting and issue with 1.6.5 that has given me a huge headache. In a nutshell, after takeoff and before starting a mission, the autopilot performs a very intense jerk which the drone tries to catch itself but frequently does not.

I can generate the bug by loading a mission on the drone and using the start button slider. It also shows up intermittently under takeoff and using the continue mission slider. I suspect the issue is either the roll axis is flipped in software and corrected mid-flight or the x/y position PID loop is running and accruing error before starting the mission. What are your thoughts?

video:
https://drive.google.com/open?id=0ByR2uOoTK3khMDgwckQyOE03eUU

log:
https://logs.px4.io/plot_app?log=49f577af-c6f9-46cb-83a7-b84590ecabf8

Thank you for any help that you can provide
Marshall

dagar, 2017-10-10:
There's a big gap in your log right before the switch to manual mode. I noticed you're using a gimbal configuration. The vmount (https://dev.px4.io/en/advanced/gimbal_control.html) driver had some very serious bugs (depending on usage) that were fixed in August.

https://github.com/PX4/Firmware/issues/7801
https://github.com/PX4/Firmware/pull/7805
https://github.com/PX4/Firmware/pull/7766
https://github.com/PX4/Firmware/pull/7844

Would you be able to try either latest PX4 master, or with a different (non gimbal) quad configuration?

DonLakeFlyer, 2017-10-11:
@mwingerson Can you include the .plan file. With that I can generate some log files from a SITL run for a similar flight plan.

DonLakeFlyer, 2017-10-11:
So in order to debug, may need a run which does not have log drop outs in it. Which may be caused by the old vmount driver chewing priority.

mwingerson, 2017-10-11:
@dagar That is good to know about the vmount. I am using that profile because it is the only hexacopter in the list that I could find. Is there a hex or quad profile that doesn't have the gimbal code? Is there a hex profile that I can switch to or do I just need to grab the latest version of PX4? 

When I opened up the GPS, I found two orientation markers on the silk screen.
[image]

Then Chris suggested comparing the pixhawk sensor orientation to what I found on the GPS. He found this diagram from here: https://pixhawk.org/sensor_orientation
[image]

This would indicate the MPU labeled cartesian coordinate is 90 degrees off of what the Pixhawk 2.1 cartesian coordinate is. Does this mean that the magnetometers in the HERE+ GPS are 90 degrees turned yaw?

[image]

With the build plate oriented as such. To fix this issue, should I just turn the GPS 90 degrees? It should be the same regardless of doing it in software or rotating the physical hardware? What would you suggest?

@Don it was a really simple mission and I didn't save the exact one. I have tested some in gazebo for kneejerk but I haven't been able to reproduce anything.

Thank you so much for your help!

dagar, 2017-10-11:
PX4 master is fairly stable right now nearing the v1.7.0 release. There should also be a generic hexa option (SYS_AUTOSTART 6001).

That GPS unit has both an HMC and LIS mag? I don't know the history, but by convention most external mags have that orientation, but the drivers automatically rotate (if on an external bus).

Through the console (FTDI or mavlink) can you run `sensors status`? 

Do you have a longer log you can share without gaps that shows EKF/mag issues?

mwingerson, 2017-10-12:
I built the latest master and setup my test rig with the new firmware. 

When I flew, I didn't get any sort of odd behavior related to kneejerk. I was seeing more GPS drops but that may have been the proximity to a tall building. I uploaded the logs to my google drive and I didn't get any video.

https://drive.google.com/drive/folders/0ByR2uOoTK3kheHplbTJnM05ibFE?usp=sharing

dagar, 2017-10-12:
Overall log isn't bad, but those GPS errors are very concerning and likely something physical (wiring, or module itself).

https://logs.px4.io/plot_app?log=7f3725bb-9961-467a-ad95-5fc24ef17fc3

[image]

"GPS module lost" is from the actual GPS driver. Problems due to proximity to tall buildings would show as poor gps quality, or EKF issues.

When you had GPS it was pretty good.

[image]

mwingerson, 2017-10-12:
you bring up very good points. Late last night, I was very happy to hope for GPS drops from the tall building. This summer I saw an instance where the GPS was getting a reset command due compass disagreement. I don't remember much on that. Do you know anything about that?

Thanks!

mwingerson, 2017-10-12:
I have these sensors values from the "sensors status" page

[image]

If I am reading this correctly, the orientation of the accelerometers agree with each other while the orientation of the magnetometers do not agree. This indicates that my orientation of the mags are incorrect. Would you agree with this assessment?

dagar, 2017-10-12:
> ... saw an instance where the GPS was getting a reset command due compass disagreement. 

Could this have been global position reset? That would be the estimator, rather than the GPS itself.

The magnitude of the compass readings in each axis are close enough that I'd try the full calibration again and compare. With everything mounted in its final place do the full 6 orientation mag calibration outside, away from buildings, cars, etc.

If you have a log from before and after then we can easily compare.

mwingerson, 2017-10-20:
Sorry for the radio silence. I have been all around testing at a high speed. I'll take a couple minutes to fill out my results and update this issue.

Quick question, is there a good way to see a log of the raw mavlink messages that are passed between QGC and the UAS?

DonLakeFlyer, 2017-10-20:
> Quick question, is there a good way to see a log of the raw mavlink messages that are passed between QGC and the UAS?

There is no logging in that direction. What part do you want to know about. If you are running a mission using Start Mission it's just a flight mode changes and arm:
```
void PX4FirmwarePlugin::startMission(Vehicle* vehicle)
{

    if (_setFlightModeAndValidate(vehicle, missionFlightMode())) {
        if (!_armVehicleAndValidate(vehicle)) {
            qgcApp()->showMessage(tr("Unable to start mission: Vehicle rejected arming."));
            return;
        }
    } else {
        qgcApp()->showMessage(tr("Unable to start mission: Vehicle not ready."));
    }
}
```

After that QGC isn't going to be sending anything to the vehicle.

dagar, 2017-10-20:
No problem, just wanted to see if the issue was resolved or not.

Depends what you mean by quick way. There are python tools you can use to dump the data (mavlogdump.py) and there's a log reply in QGC, but it's fairly broken.

DonLakeFlyer, 2017-10-20:
* UAS->QGC yes, logging is available from QGC. Log Replay in daily builds is working now. I've been working on it. On desktop builds telemetry logs are on by default and store int he QGroundControl/Telemetry direction.
* QGC->UAS no logging available.

mwingerson, 2017-10-23:
Hey guys,

I got a set of trials here for you to checkout. This is the set of logs for my entire flight out.

You should see alternating tests (20 total) with the same UAS and alternating the latest code from mavlink QGC and the latest that we have in our private repository. Success rate is defined by complete flight without GPS drop (which we shouldn't be seeing).  

100% success on QGC from mavlink branch
20% success on QGC from OKSI branch

Mavlink Branch QGC:
 - https://logs.px4.io/plot_app?log=0df4826f-1711-4b71-b0b9-ff0106b08558

OKSI Branch QGC:
 - https://logs.px4.io/plot_app?log=76e5b7f6-dc40-47dd-bcb0-a5698b13bf45

All the logs:
 - https://drive.google.com/open?id=0ByR2uOoTK3khMXR3M0lXUXNJcmc

I recorded video from the ground. Do you need me to upload that?

Thanks!

DonLakeFlyer, 2017-10-23:
> Mavlink Branch QGC

This is which build. Daily/Stable? Built it yourself or downloaded?

mwingerson, 2017-10-23:
Mavlink master branch commit: e842fd7a5dcd2f9cb7171553dde8e336c93630e6

OSKI master branch commit: 2a5940d0d7d91cbad6d3a95f698a9d851ac3daa1

I cloned both from their respective repos and built locally

DonLakeFlyer, 2017-10-24:
I just verified that the StartMission code in QGC for both those codebases are exactly the same. So I'm at a loss as to understanding how QGC could be affecting GPS drop.

mwingerson, 2017-10-26:
Any update on this Don? It is clearly an issue with our branch of QGC, right? Let me know how I can help you find a fix to this problem.

DonLakeFlyer, 2017-10-26:
I have no idea how QGC could affect the GPS. If you are running a mission the code to start the mission is exactly the same in both code bases.

@LorenzMeier I really need help here in flight log analysis.

mwingerson, 2017-10-26:
@DonLakeFlyer I am also at a loss for how QGC could be generating this kind of behavior without a deliberate task that affects the operating parameters of the UAS. What have you tested so far? 

@LorenzMeier I would also like to know more about log analysis. What programs do you use? I have been using the PX4 online log reader and it seems good but I think I am just interested in more than is just showed there.

DonLakeFlyer, 2017-10-26:
Can you post a .pan file so I can try to create a mission to fly that is as close to yours as possible?

DonLakeFlyer, 2017-10-26:
Also what version of firmware are you flying?

mwingerson, 2017-10-26:
Here is a plan file:
https://drive.google.com/open?id=0ByR2uOoTK3khZUowTDh0ZkFxaG8

I'm on firmware commit: bed4714cfe3fe43c785236a9db625014c7781f31

Let me know if there is anything else and thanks!

DonLakeFlyer, 2017-10-26:
Fly my Iris on firmware master using OKSI QGC. Tried manual flight, missions from Start Mission and so forth. All worked fine.

LorenzMeier, 2017-10-26:
@mwingerson I'm looking at this.

@priseborough Could you please chime in?

LorenzMeier, 2017-10-26:
@mwingerson Do you have MAVLink log streaming enabled on your QGC build? Please try disabling it.

@bkueng Could you have a look at this log? The log system behaviour doesn't look quite right.
https://logs.px4.io/plot_app?log=76e5b7f6-dc40-47dd-bcb0-a5698b13bf45

priseborough, 2017-10-26:
I am not seeing estimator time slips, so the imu, sensor and ekf2 modules are running at the right rate , ekf innovations are low and the cpu load is stable. 

There have been some reported instances in ArduPilot of the mag in the Here GPS units not starting correctly on some setups and going to an incorrect configuration due to a floating supply voltage when powered down for short periods between tests (it has 200uF of supply capacitance). You could try connecting a 1K resistor between GND and 5V on the GPS module connector.

bkueng, 2017-10-27:
@mwingerson you can use FlightPlot for further log analysis: https://dev.px4.io/en/log/flight_log_analysis.html

What I see is this: QGC tries to enable log streaming, but it fails: logger is waiting for an ack from QGC, but it never gets it and times out. During this time the logger waits by polling on the ack topic, which explains the logging drops in the beginning. The system continues to run normally during that time, which becomes apparent by the estimator timeslip not increasing and the absence of any sensor timeouts. From what I see, this cannot affect flight behavior, the logger and mavlink run at lower priority and there is no busy-loop or priority inversion.
The only message I cannot fully explain is `WARN  [mavlink] no ack from logger (is it running?)`. It means the mavlink module did not get an ack from the logger within 300ms. But there's also `[logger] Ack timeout. Stopping mavlink log`, meaning the logger did get the request, and published the command ack, and mavlink did not receive it. I see 2 potential causes:
- not enough FD's: possible, since the logs show an FW version before the `orb_exists` optimization (https://github.com/PX4/Firmware/pull/8135)
- dropped ack. Even though it uses a queue of 3 elements, the mavlink uorb subscriptions use `orb_stat` to check for message updates, which returns the time of the newest element in the queue. And during automated takeoff, there are multiple commands sent, so I see it possible that an ack gets dropped (or rather: it's not dropped, but mavlink does not see that there are multiple ack's in the queue, only the first one).

I could not reproduce this so far.

@DonLakeFlyer does QGC try to enable log streaming on low-bandwidth telemetry links too? If so, it should not do that.

DonLakeFlyer, 2017-10-27:
> does QGC try to enable log streaming on low-bandwidth telemetry links too? If so, it should not do that.

@dogmaphobic You'll have to ask Gus about that. I don't know much about that code.

dogmaphobic, 2017-10-27:
Log streaming is disabled by default, regardless the link type. You must manually enabled it:

[image]

If you manually enable it, currently there are no checks related to link bandwidth and/or suitability.

bkueng, 2017-10-30:
Ok, I added the check on the vehicle side: #8209

DonLakeFlyer, 2017-10-30:
@mwingerson Has turning off mavlink log streaming fixed the problem?

mwingerson, 2017-10-30:
I went to test quickly yesterday morning and ran 2 reps with logging off without any problem. Then I turned logging back on and cratered my test rig. I fixed up the test rig and successfully ran 15 reps without a problem. Since this first crash was so violent, I was worried about testing it again.

I am in LA now and I will be testing in an interleaved fashion as before on my most reliable S900 frame. I'll let you know the results.

Thanks for all the help!

DonLakeFlyer, 2017-10-30:
Is there a reason you need to turn on/use mavlink streaming? You can download vehicle logs after the fact if you want.

dogmaphobic, 2017-10-31:
If the drone falls into the Mariana Trench, you may have some issues trying to get to the logs within the autopilot :P

DonLakeFlyer, 2017-10-31:
Heehee. Guess it would be good to get it working.

mwingerson, 2017-10-31:
Good news! Turning off the mavlink 2.0 logger has eliminated the problem. At some point, I remember my boss asking me to turn on all logging that I can find. So I am sure I turned on the MAVlink 2.0 logging for my next flight test and didn't associate the kneejerk behavior with the change. 

As for why our branch had this issue and the others didn't, that setting is saved in the user configs and our branch is named different so it saves the settings in a different directory for the MAVlink repo. 

Can anybody explain a solid mechanism for this behavior? I suspect it is something to do with QGC either sending or requesting too much data for the drone mid-flight. Since the drone is trying to service the logger, it bogs down or hangs. Then something kicks the drone back into service, like a watchdog timer. At this point, the drone is mid-air and trying to catch up with lost time causing an over-reaction like I was seeing. Any thoughts?

Thanks again for all the help!

DonLakeFlyer, 2017-10-31:
@dogmaphobic Isn't this a streaming push protocol which the vehicle side is in control of?

> What I see is this: QGC tries to enable log streaming, but it fails: logger is waiting for an ack from QGC, but it never gets it and times out. During this time the logger waits by polling on the ack topic, which explains the logging drops in the beginning.

Also could you double check this?

dogmaphobic, 2017-10-31:
The only message sent to the vehicle is to *start* streaming. From that point on, QGC only receives log messages.

mwingerson, 2017-10-31:
another thought, I had to switch to MAVlink 1.0 because I was having issues with uploading missions in MAVlink 2.0.  Does this shed some light?

DonLakeFlyer, 2017-10-31:
> I had to switch to MAVlink 1.0 because I was having issues with uploading missions in MAVlink 2.0.

Huh? Can you make sure to enter a seperate issue on that?

mwingerson, 2017-10-31:
I tend to not release details like that until I have tested more definitively. I will put that on my to do list to verify the result and open an issue on it.

DonLakeFlyer, 2017-10-31:
For that one, the important thing to note is the QGC, firmware and Mavlink versions. Mission Protocol has changed in the newest versions of both in order to add support for GeoFence/Rally. There were a couple quirks but I think it's all fixed on master of everything.

closed 2018-02-21 by PX4BuildBot