PX4/PX4-Autopilot #8109: Quadrotor having vertical Z up velocity in Altitude hold mode

siddyyagno3, 2017-10-11:
The Quadrotor has a optical flow px4 flow and Lidar lite V3(i2c). While flying manually the drone is performing normally but when switching to Altitude Hold mode. It starts going up and I have to take manual control. I am using EKF2 in gps denied environment. I also tried changing the aid_mask as well as recalibrating but same. 
Should I change to LPE for better output or I am setting some parameter wrong so that distance sensor data is not fusing? Anyway I checked the lidar reading in analyzer topic current distance they are consistent with the actual height. In flight we never went above 4m which is not visible as per local position z. Can this be related to hardware setup?
Please suggest the best setting for a good hover with my current set up.
Flight logs:
[https://logs.px4.io/plot_app?log=549f5587-f496-4e81-acb5-ca4b23b629c1](url)

[https://logs.px4.io/plot_app?log=1dd97b1d-0ba3-42e2-9e72-05ed7219191d](url)

AlexisTM, 2017-10-12:
Hey there:

In both graphs, the Z position is going up and then is reset, while the Z measured distance is correct. Could you try to **disable the barometer**? 

https://logs.px4.io/plot_app?log=1dd97b1d-0ba3-42e2-9e72-05ed7219191d#Nav-Local-Position-Z

https://logs.px4.io/plot_app?log=549f5587-f496-4e81-acb5-ca4b23b629c1#Nav-Local-Position-Z

EKF reports : `[lib__ecl] EKF rng hgt timeout - reset to rng hgt`

@ChristophTobler & @priseborough

mad-sci, 2017-10-12:
Hello @AlexisTM 

Below are the parameters related to BARO performance. Which one would you recommend changing, BARO_NOISE or BETA_NOISE be increased or decreased? Or is there any other parameter which also needs modification.

EKF2_BARO_DELAY, 0.0
EKF2_BARO_GATE, 5.0
EKF2_BARO_NOISE, 5.0
EKF2_BETA_NOISE, 0.30000001192092896

ChristophTobler, 2017-10-13:
@siddyyagno3 It looks like your Hardware is not dampened correctly. Can you send a picture of your hardware? Have a look at your acceleration -> https://logs.px4.io/plot_app?log=1dd97b1d-0ba3-42e2-9e72-05ed7219191d#Nav-Raw-Acceleration. The ekf2's height estimation is quite sensible to noisy IMU data. Can you try fixing your dampening and check if it still happens?

ChristophTobler, 2017-10-13:
This has nothing to do with barometer or barometer parameters as `EKF2_HGT_MODE` was set to range_fidner (2). Therefore it uses the distance sensor for height estimation and not the barometer.

siddyyagno3, 2017-10-13:
@ChristophTobler 
It looks like that may be the issue but I also had that intuition that is why I also tested a previous flight with EKF2_aid_mask set to inhibit IMU bias but the result was same. Doesn't this parameter settings takes IMU acceleration out of the picture?
I don't have flight data to support that. I will redo the same flight. Subsequently, 
@mad-sci we will improve the damping and lower the vibration and upload the test flight data as the raw acceleration do look very bad in our case.

ChristophTobler, 2017-10-13:
@siddyyagno3 Unfortunately there is no parameter that magically solves mechanical noise/issues ;)
Did you ever do a "normal" flight outdoors using just GPS and barometer? Would be surprised if that works.

priseborough, 2017-10-13:
I agree with @ChristophTobler. Your vibration levels are much too high. You need to isolate your autopilot board from the airframe using soft mounts and make sure your connecting cables have adequate slack and bend in them so they also do not transmit vibration. 

This is not a firmware problem.

closed 2017-10-13 by priseborough

mad-sci, 2017-10-13:
@ChristophTobler @AlexisTM @priseborough Thank you for your inputs and got better results after changing the dampers of the Pixhawk. Still the local position Z (https://logs.px4.io/plot_app?log=ebf3160c-8049-4971-b362-f761db190f2e#Nav-Local-Position-Z)is not good enough as in this (https://logs.px4.io/plot_app?log=88e4ba1c-8f73-4b66-b988-c7cbd34f7739#Nav-Local-Position-Z)
Do we need to change any other parameters?

My Flight logs https://logs.px4.io/plot_app?log=ebf3160c-8049-4971-b362-f761db190f2e

ranjeetkumar16722, 2019-08-05:
In altitude hold mode, vehicle will ascend to the height specified by parameter MIS_LTRMIN_ALT and hold at that altitude. Please check the parameter what you have set.