PX4/PX4-Autopilot #8116: RTL naming convention

sanderux, 2017-10-12:
In reference to https://github.com/mavlink/qgroundcontrol/issues/5725 we need to find consensus on the naming of what we now call RTL.

**Return to Land**
Perhaps the most common definition of this behavior but slightly ambiguous and sometimes inaccurate as RTL does not imply it will actually Land.

**Return to Launch**
Also used in several places, this gets closer to what usually happens as the home position is set to the launch point, but this could be updated during the flight.

**Return to Home**
Perhaps the most accurate description of what the autopilot is instructed to do, but this would change the abbreviation that has become common use (RTL would become RTH). 

Another thing to consider when we start implementing Safe Landing Spots (RallyPoints) is that, if we implement this behavior in the same function as RTL, every name in the list above no longer covers the actual behavior. For this and various other reasons we could consider implementing a different flight mode that specifically covers this behavior (failsafe flight directly to home, destination, or a failsafe location, which-ever is closest). 

I propose we discuss this in the next dev call.

FYI @hamishwillee @dagar @Antiheavy @DonLakeFlyer @LorenzMeier

dagar, 2017-10-12:
Need to consider where future additions fit. How far do we overload RTL or do we introduce more modes.
 - Land at nearest safe landing spot (https://github.com/PX4/Firmware/pull/7743)
 - RTL -> land using mission (DO_LAND_START) (https://github.com/PX4/Firmware/pull/7363)
 - SmartRTL - return via recorded flight path (https://github.com/PX4/Firmware/pull/7363)

There's also the configuration of the existing RTL.
 - CLIMB -> RETURN -> DESCEND (configurable) -> LOITER (configurable) -> LAND (optional)

hamishwillee, 2017-10-12:
@mrpollo Can this topic be added to the devcall agenda?

I don't think the mode naming can be agreed until @dagar point is addressed - whether there are a bunch of modes or just one. If just one mode, then a name like "Return to Safety", "Rally", "Muster", "Regroup" might work, where the safe point is defined as home, launch point or whatever.

sanderux, 2017-10-13:
In my opinion we should separate general RTL behavior from specific RTL behavior.

RTL using DO_LAND_START and SmartRTL i would consider general RTL behavior settings that should be set by either parameters or by the fact of having them programmed.

Failsafe landing i consider specific RTL behavior to which i would not even find it necessary to call it RTL. I would opt we create a new flightmode so the user can control what event should constitute what behavior. EG: flicking RTL manually or radio loss should constitute a normal RTL whereas Battery failsafe or quadchute should constitute failsafe landing.

Antiheavy, 2017-10-13:
> Failsafe landing i consider specific RTL behavior to which i would not even find it necessary to call it RTL.

i agree, specific things like Loss of GPS failsafe (NAV_GPSF_xx), emergency battery level failsafe (BAT_EMERGEN_THR), etc are not part of Return modes and should have their own specific emergency land behavior (for the most part do now I think) .  The "failsafe" concept is really just "how does the system 'act' (i.e. take automatic action) if X issue occurs".  Executing a Return mode or executing an emergency land type behavior are just two categories of actions.  As you say, if this discussion is focused on "RTL naming conventions" we should be careful not to conflate "failsafe".

Antiheavy, 2017-10-13:
My recent thinking on the naming convention problem is to use just the word "Return" when talking about this generically and start thinking about the different the RTH, RTL, SmartRTH, ReturntoSafePoint, etc as user selectable sub-modes (or maybe just sub-behaviors) that define the "Return" behavior.

I need a picture, in my mind this is a logical flow:

[image]

Antiheavy, 2017-10-13:
Here is a hypothetical way the above Return behaviors might map out to a consistent user interface in QGC.  Wording probably needs tweaking, but in general the various failsafe options would all just say "Return" and the "Return Home Settings" changes to "Return Settings":
[image]

hamishwillee, 2017-10-14:
Of course in some cases you're not actually returning, but that aside this looks workable. What I was going to suggest was seeing if we could get some UX expertise on this.

tubeme, 2017-10-16:
This was a long time ago discussion (2 years ago). And then we agreed that we will call all this conditions "Return" as what most people experienced and inexperienced will relate to the meaning of the word. Also this was part of the user experience and positioning etc.

Now it is called "Return" in the QGC. I cannot remember what the voice message was. May be the voice message is still "Return to launch activated".

The three modes @dagar is proposing could be:

Land at nearest safe landing spot (#7743) --  "Rally", "To Rally Point"
RTL -> land using mission (DO_LAND_START) (#7363) - "Return"
SmartRTL - return via recorded flight path (#7363) - "Smart Return"

dagar, 2017-10-16:
Is rally a common term? To me it definitely doesn't directly map to the idea of finding the nearest safe spot to land.

This is difficult because I want to be careful not to expose more complexity to the user than needed.

Antiheavy, 2017-10-16:
Part of the problem is that the text/voice messages, QGC metadata, and mode names aren't consistent about the use of the word "Return" and if or what additional words follow.

To add to the confusion, for example, there is also a mode (or is it "navigation state"?) called "Return to Groundstation"
https://github.com/PX4/Firmware/blob/3c18be387cc47c9875137745b171e7ccec901f2e/msg/vehicle_status.msg#L23

Antiheavy, 2017-10-16:
> Is rally a common term? To me it definitely doesn't directly map to the idea of finding the nearest safe spot to land.

Sorry if I wander slightly off-topic:

There was a commercial autopilot system that was very popular a decade ago that was the first (that I know of) to start using the term "Rally" for a special loiter location that could be placed somewhere.  It was often used as a "mission pause" location or "safe hold" or "Safe divert" or whatever.  It was sometimes used as the hold or gathering spot(s) when flying UAV "swarms".  It also functioned as a 2nd location different from Home where you could do Home type things.  e.g. if you lost comms you could set the system to return to the Rally point instead of the Home point, and then optionally land.  For fixed wing it became common practice to use the Rally as the "loiter down" waypoint before breaking out on a glide landing.

I feel like the APM/MisisonPlanner folks have been using the term Rally recently with a slightly different meaning, focused more on "safe landing spots", but I'm not exactly sure the full thinking.

For what it is worth here is webster's:  https://www.merriam-webster.com/dictionary/rally

To me the word Rally does not seem a good choice of word if "Safe Landing Spot" is the only intended function.  The word "Return" still works for me in this context as you are returning to the ground (if you're an aircraft).  However, maybe there between the various MAVlink systems there is a new emerging use and definition of the term Rally for this purpose.  If so, if that is the case then maybe "Rally" should be it's own thing and not really associated with "Return".

tubeme, 2017-10-16:
@dagar You said it "Safe Spot Land" mode. 

The rally point landing could not go under "Return" label...

JonReacher, 2017-10-16:
I very much agree with Antiheavy and the Return only naming, then the ability to define return to where and do what. Return to Land can be confusing at first because does that mean "return home in order to land" or "return to the earth, now, directly underneath where it is when triggered"? Land can be a place or an action.
 I also like Safe Spot Land. Seems intuitive.

sanderux, 2017-10-16:
I think we should have a different mode for regular rtl and failsafe rtl. The manual switch to return should not go to a failsafe spot, neither does geofence or rc loss.

Battery failsafe should not.

tubeme, 2017-10-16:
@sanderux It could be "Failsafe Return"

hamishwillee, 2017-10-16:
"Safe Landing Point" implies landing, and we're not always landing. "Return" is not always what we are doing. 
"Safe Recover Mode" - I quite like because we don't imply return or land, but we do imply movement and safety.

Antiheavy, 2017-10-16:
> I think we should have a different mode for regular rtl and failsafe rtl. The manual switch to return should not go to a failsafe spot, neither does geofence or rc loss.
> 
> Battery failsafe should not.

Took me a while to understand what you mean, but yes, I think follow you now.  Please correct me if I am wrong, but are you saying that "Safe Spot Land" or "Rally" or "Failsafe RTL" or "Failsafe Return" (all different names for the same thing) should not be part of "Return" mode and should be a different mode?  I think a loose consensus is forming around that idea in the issue discussion.

Antiheavy, 2017-10-16:
So here is a different way to look at "Return" mode that removes the "Safe Point" (or whatever it gets called) idea.
[image]

Antiheavy, 2017-10-16:
> Battery failsafe should not.

Low/Critical/Emergency battery failsafe has it's own confusing lack of consistent naming conventions and could be another issue post all in itself.   If "Safe Point" (or whatever it gets called) is implemented I could see that becoming one of the failsafe options for one of the stages of battery depletion (i.e. a new value for COM_LOW_BAT_ACT).  "Return" mode should still be a valid failsafe option for some of the stages of battery depletion (here I am talking about Return mode as I've diagrammed immediately above).

sanderux, 2017-10-16:
Yes indeed.

Return = "Go back to where you took off" 
This is desired behavior for stuff like geofence, radio loss or manual engagement of return switch
path tracking could be a setting on this behavior

Failsafe landing = "Land at home, destination or failsafe point, whichever is closest"
This is desired behavior at low battery, quadchute, engine failure, etc

tubeme, 2017-10-16:
I think just Return is good enough. 

We can have the voice pronounce the reason for return...  and we have a message pop up in QGC if it is Battery Failsafe, Failsafe, comanded or other type of Return.

The Rally point mode could be just "Safe Spot" mode.

Keep it simple is not a bad idea.

tubeme, 2017-10-16:
We could introduce the term emergency. Emergency Landing instead of Failsafe Land.

Emergency Return. Emergency Spot Land.

If we want to be more understandable for the novice and more intuitive for the advanced users.

Antiheavy, 2017-10-16:
> I think just Return is good enough.
> 
> We can have the voice pronounce the reason for return... and we have a message pop up in QGC if it is Battery Failsafe, Failsafe, comanded or other type of Return.
> 
> The Rally point mode could be just "Safe Spot" mode.
> 
> Keep it simple is not a bad idea.

I agree with this.

Antiheavy, 2017-10-16:
I now notice @sanderux was advocating that Rally/Safe Spots should be a separate mode from RTL (or Return) a few months ago  ;-)  https://github.com/PX4/Firmware/pull/7743   I'm coming around to this idea too.

sanderux, 2017-10-16:
We'll discuss this in the dev call Wednesday. Feel free to join in

Antiheavy, 2017-10-30:
Thanks for inviting me to participate in the dev call the other week.  My understanding of the consensus reached by the group who discussed this topic is summarized as:

1) "Return" mode may have several sub-varients such as "smart return" and "return to land / return to mission landing", etc.  These fall under the generic category of Return.  sub-varients may have their own specific messages and ways of informing the users, but the top level "mode" is still just "Return".

2) Rally/Safe spots should not be part of Return.  It should be it's own mode so that may be separately selected by the user for some failsafes (e.g. low/critical battery) and at the same time be able to select "Return" mode for other failsafes (e.g. loss of datalink).

hamishwillee, 2017-10-31:
@Antiheavy Thanks for that information/clarification.

> Thanks for inviting me to participate in the dev call the other week. 

Just FYI, I've heard anecdotally that some people have assumed that dev calls are for the "core dev team". In fact they are for every interested/serious developer with something useful to contribute.  You can attend anytime and will be very welcome. If you have a specific topic to discuss, just make sure it gets added to the agenda.

closed 2018-02-21 by PX4BuildBot