PX4/PX4-Autopilot #8120: **Feature Request** Bring back gimbal stabilisation feature for servo gimbals

mje-nz, 2017-10-12:
If I understand things correctly, before #5389 the vmount module (called gimbal at the time) was used to stabilize a gimbal, and now it is used for passthrough control of gimbals (now including Mavlink gimbals). The stabilization code is still in there, but the only way I can see to activate it is to through MAV_CMD_DO_MOUNT_CONFIGURE.

I'm interested in this module for use with a servo gimbal. Another issue for that application is that the default output range is 360 degrees; it would be nice if that was adjustable without having to edit the mixers so that stabilization roughly works without too much effort. The Ardupilot implementation of this feature has the PWM limits and the angle limits adjustable from the GCS, but that's probably going a bit far here.

I propose adding a parameter MNT_STABILIZE to control whether stabilization is on or off by default, and MNT_RANGE_PITCH, MNT_RANGE_ROLL, and MNT_RANGE_YAW to set the output range. Does that seem reasonable and would you accept a PR?

dagar, 2017-10-13:
Sounds reasonable to me if you're able to implement and test the PR. Once the configuration is exposed as params the last step of making a gui to edit them in QGC is pretty easy.

mje-nz, 2017-10-25:
Implemented in #8149

closed 2017-10-25 by mje-nz