PX4/PX4-Autopilot #8121: mavros cannot connect the px4 when sitl in gazebo

YuYangguang, 2017-10-13:
My work environment is ubuntu 14.04+ gazebo7+ros-jade

when I follow the instruction in https://dev.px4.io/en/simulation/multi-vehicle-simulation.html and launch the file multi_uav_mavros_sitl.launch. I can see two simulated quad. But the mavros cannot connect the px4. Then  I check the udp usage in my computer, I found only the 14560 port, which is the port for communication between the gazebo and px4. The udp ports 14556 and 14557, which is port for communication between the mavros and px4,  are not in use. Is there anything wrong? Who can help me?

dagar, 2017-10-17:
@TSC21 @hamishwillee any suggestions?

hamishwillee, 2017-10-17:
@dagar Not really - I haven't tried the multivehicle case and I have only tried (anything) on latest Ubuntu.
Looking at the docs I would expect ROS to connect of 14557 and [14559](https://github.com/PX4/Firmware/blob/master/launch/multi_uav_mavros_sitl.launch#L65)

@mzahana did a commit on this doc - and may have some ideas.

mzahana, 2017-10-18:
@YuYangguang Do you have the ```iris_1``` & ```iris_2```  files in your Firmware folder? You can find them in ```Firmware/posix-configs/SITL/init/ekf2``` and ```Firmware/posix-configs/SITL/init/lpe```

They hold the commands that start the mavlink UDP ports
see [this](https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/ekf2/iris_1#L67) and [this](https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/ekf2/iris_2#L67)

closed 2018-02-21 by PX4BuildBot

zubair123khan, 2018-06-16:
@mzahana I am using px4 mavros_posix_sitl.launch. But in rqt_graph too i am not able to see any connection with my drone and mavros. Any suggestions ?? Thanks