PX4/PX4-Autopilot #8122: Param generation broken

hamishwillee, 2017-10-13:
**Bug Report**

I'm trying to run `px_process_params.py -m` to regenerate parameter docs (I think they may be out of sync). ```
Firmware\Tools>python px_process_params.py -m -v
```
Problem is that this reports `Warning: no parameters found`

I think that it may have been broken by this commit: https://github.com/PX4/Firmware/commit/db816982cde07a19975b02bd2e7d63242d5d8f04#diff-cf82e6d68ab6995af20261d2c826aae0

See that the old version walks the tree, generating file lists for all nested files. The new version just lists the files in the current directory. 

FYI @dagar - let me know if you want me to try fix.

dagar, 2017-10-17:
I'm wondering if this might be a good excuse to rename the param files to something identifiable (.param or .px4param, etc). Right now they're all named .c, but not actually source code anymore.
Then in cases like this we can find all *.px4param and run the script.

dagar, 2017-10-17:
Unless an update to the current parameter listing is badly needed I'd like incorporate some overdue improvements to the param generation and fix this issue in the process. https://github.com/PX4/Firmware/issues/8141

hamishwillee, 2017-10-17:
OK, as long as this isn't a "months" process. Irrespective of all other discussions, I think it makes real sense to rename the files `*.param` or similar

hamishwillee, 2017-11-03:
Closing as @dagar has automated parameter generation now.

closed 2017-11-03 by hamishwillee