PX4/PX4-Autopilot #8124: Cannot build firmware on Pixhawk Mini

elbon1, 2017-10-13:
I currently have Ubuntu 16.04.3 LTS dual-boot with Windows 10 and I have followed the PX4 Developer Guide. I tried to build the firmware on Pixhawk Mini by using `make px4fmu-v2_default` and I'm getting the following output:

`[10/387] Building CXX object src/modul...ient.dir/microRTPS_client_main.cpp.obj
FAILED: /usr/bin/arm-none-eabi-g++   -DBUILD_URI=localhost -DCONFIG_ARCH_BOARD_PX4FMU_V3 -DMODULE_NAME=\"micrortps_client\" -DPX4_MAIN=micrortps_client_app_main -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I. -Isrc -Isrc/modules -I../../src -I../../src/drivers/boards/px4fmu-v3 -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/nuttx/include -INuttX/nuttx/include/cxx -I../../platforms/nuttx/NuttX/apps/include -Iexternal/Install/include -I../../src/modules/micrortps_bridge/micrortps_client -Isrc/modules/micrortps_bridge/micrortps_client/micrortps_client -I../../src/lib/micro-CDR/include -Isrc/lib/micro-CDR/include/microcdr -fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wreorder -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Os -DNDEBUG   -Wframe-larger-than=4096 -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -MMD -MT src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/microRTPS_client_main.cpp.obj -MF src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/microRTPS_client_main.cpp.obj.d -o src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/microRTPS_client_main.cpp.obj -c ../../src/modules/micrortps_bridge/micrortps_client/microRTPS_client_main.cpp
In file included from /usr/arm-none-eabi/include/c++/6.3.1/string:52:0,
                 from src/modules/micrortps_bridge/micrortps_client/micrortps_client/microRTPS_transport.h:35,
                 from ../../src/modules/micrortps_bridge/micrortps_client/microRTPS_client_main.cpp:33:
/usr/arm-none-eabi/include/c++/6.3.1/bits/basic_string.h: In function 'int std::__cxx11::stoi(const wstring&, size_t*, int)':
/usr/arm-none-eabi/include/c++/6.3.1/bits/basic_string.h:5534:42: error: 'wcstol' is not a member of 'std'
   { return __gnu_cxx::__stoa<long, int>(&std::wcstol, "stoi", __str.c_str(),
                                          ^~~
compilation terminated due to -Wfatal-errors.
[10/387] Building CXX object src/modul...rortps_client/microRTPS_client.cpp.obj
FAILED: /usr/bin/arm-none-eabi-g++   -DBUILD_URI=localhost -DCONFIG_ARCH_BOARD_PX4FMU_V3 -DMODULE_NAME=\"micrortps_client\" -DPX4_MAIN=micrortps_client_app_main -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I. -Isrc -Isrc/modules -I../../src -I../../src/drivers/boards/px4fmu-v3 -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/nuttx/include -INuttX/nuttx/include/cxx -I../../platforms/nuttx/NuttX/apps/include -Iexternal/Install/include -I../../src/modules/micrortps_bridge/micrortps_client -Isrc/modules/micrortps_bridge/micrortps_client/micrortps_client -I../../src/lib/micro-CDR/include -Isrc/lib/micro-CDR/include/microcdr -fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wreorder -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Os -DNDEBUG   -Wframe-larger-than=4096 -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -MMD -MT src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/micrortps_client/microRTPS_client.cpp.obj -MF src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/micrortps_client/microRTPS_client.cpp.obj.d -o src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/micrortps_client/microRTPS_client.cpp.obj -c src/modules/micrortps_bridge/micrortps_client/micrortps_client/microRTPS_client.cpp
In file included from /usr/arm-none-eabi/include/c++/6.3.1/string:52:0,
                 from src/modules/micrortps_bridge/micrortps_client/micrortps_client/microRTPS_transport.h:35,
                 from src/modules/micrortps_bridge/micrortps_client/micrortps_client/microRTPS_client.cpp:33:
/usr/arm-none-eabi/include/c++/6.3.1/bits/basic_string.h: In function 'int std::__cxx11::stoi(const wstring&, size_t*, int)':
/usr/arm-none-eabi/include/c++/6.3.1/bits/basic_string.h:5534:42: error: 'wcstol' is not a member of 'std'
   { return __gnu_cxx::__stoa<long, int>(&std::wcstol, "stoi", __str.c_str(),
                                          ^~~
compilation terminated due to -Wfatal-errors.
[10/387] Building CXX object src/modul...tps_client/microRTPS_transport.cpp.obj
FAILED: /usr/bin/arm-none-eabi-g++   -DBUILD_URI=localhost -DCONFIG_ARCH_BOARD_PX4FMU_V3 -DMODULE_NAME=\"micrortps_client\" -DPX4_MAIN=micrortps_client_app_main -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I. -Isrc -Isrc/modules -I../../src -I../../src/drivers/boards/px4fmu-v3 -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/nuttx/include -INuttX/nuttx/include/cxx -I../../platforms/nuttx/NuttX/apps/include -Iexternal/Install/include -I../../src/modules/micrortps_bridge/micrortps_client -Isrc/modules/micrortps_bridge/micrortps_client/micrortps_client -I../../src/lib/micro-CDR/include -Isrc/lib/micro-CDR/include/microcdr -fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wreorder -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Os -DNDEBUG   -Wframe-larger-than=4096 -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-opt [... cut ...]
/usr/arm-none-eabi/include/c++/6.3.1/bits/basic_string.h: In function 'int std::__cxx11::stoi(const wstring&, size_t*, int)':
/usr/arm-none-eabi/include/c++/6.3.1/bits/basic_string.h:5534:42: error: 'wcstol' is not a member of 'std'
   { return __gnu_cxx::__stoa<long, int>(&std::wcstol, "stoi", __str.c_str(),
                                          ^~~
compilation terminated due to -Wfatal-errors.
[10/387] Building CXX object src/modul...les/modules__logger.dir/logger.cpp.obj
ninja: build stopped: subcommand failed.
Makefile:155: recipe for target 'px4fmu-v3_default' failed
make: *** [px4fmu-v3_default] Error 1
`

dagar, 2017-10-17:
Hi @elbon1 

The output is a bit hard to follow like that, but could you first make sure your dev environment is set up to match the current documentation (https://dev.px4.io/en/setup/dev_env_linux.html) and you're using the latest PX4 Firmware master?

If it's still failing please post the error in a code block or to a service like pastebin.

elbon1, 2017-10-17:
I'm sorry, I'm still new to this. Yes I set up the dev environment from that link, by downloading **ubuntu_sim.sh** and running it on the terminal. The PX4 Firmware master is installed from there, hence I expect to have the latest version. Also, when I built the jMAVSim target, the simulator ran as expected. I re-run `make px4fmu-v3_default` and the error is as follows: 

```ruby
[6/361] Building CXX object src/module...tps_client/microRTPS_transport.cpp.obj
FAILED: /usr/bin/arm-none-eabi-g++   -DBUILD_URI=localhost -DCONFIG_ARCH_BOARD_PX4FMU_V3 -DMODULE_NAME=\"micrortps_client\" -DPX4_MAIN=micrortps_client_app_main -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I. -Isrc -Isrc/modules -I../../src -I../../src/drivers/boards/px4fmu-v3 -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/nuttx/include -INuttX/nuttx/include/cxx -I../../platforms/nuttx/NuttX/apps/include -Iexternal/Install/include -I../../src/modules/micrortps_bridge/micrortps_client -Isrc/modules/micrortps_bridge/micrortps_client/micrortps_client -I../../src/lib/micro-CDR/include -Isrc/lib/micro-CDR/include/microcdr -fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wreorder -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Os -DNDEBUG   -Wframe-larger-than=4096 -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -MMD -MT src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/micrortps_client/microRTPS_transport.cpp.obj -MF src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/micrortps_client/microRTPS_transport.cpp.obj.d -o src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/micrortps_client/microRTPS_transport.cpp.obj -c src/modules/micrortps_bridge/micrortps_client/micrortps_client/microRTPS_transport.cpp
In file included from /usr/arm-none-eabi/include/c++/6.3.1/string:52:0,
                 from src/modules/micrortps_bridge/micrortps_client/micrortps_client/microRTPS_transport.h:35,
                 from src/modules/micrortps_bridge/micrortps_client/micrortps_client/microRTPS_transport.cpp:39:
/usr/arm-none-eabi/include/c++/6.3.1/bits/basic_string.h: In function 'int std::__cxx11::stoi(const wstring&, size_t*, int)':
/usr/arm-none-eabi/include/c++/6.3.1/bits/basic_string.h:5534:42: error: 'wcstol' is not a member of 'std'
   { return __gnu_cxx::__stoa<long, int>(&std::wcstol, "stoi", __str.c_str(),
                                          ^~~
compilation terminated due to -Wfatal-errors.
[6/361] Building CXX object src/module...ient.dir/microRTPS_client_main.cpp.obj
FAILED: /usr/bin/arm-none-eabi-g++   -DBUILD_URI=localhost -DCONFIG_ARCH_BOARD_PX4FMU_V3 -DMODULE_NAME=\"micrortps_client\" -DPX4_MAIN=micrortps_client_app_main -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I. -Isrc -Isrc/modules -I../../src -I../../src/drivers/boards/px4fmu-v3 -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/nuttx/include -INuttX/nuttx/include/cxx -I../../platforms/nuttx/NuttX/apps/include -Iexternal/Install/include -I../../src/modules/micrortps_bridge/micrortps_client -Isrc/modules/micrortps_bridge/micrortps_client/micrortps_client -I../../src/lib/micro-CDR/include -Isrc/lib/micro-CDR/include/microcdr -fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wreorder -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Os -DNDEBUG   -Wframe-larger-than=4096 -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-o [... cut ...]
/usr/arm-none-eabi/include/c++/6.3.1/bits/basic_string.h: In function 'int std::__cxx11::stoi(const wstring&, size_t*, int)':
/usr/arm-none-eabi/include/c++/6.3.1/bits/basic_string.h:5534:42: error: 'wcstol' is not a member of 'std'
   { return __gnu_cxx::__stoa<long, int>(&std::wcstol, "stoi", __str.c_str(),
                                          ^~~
compilation terminated due to -Wfatal-errors.
[6/361] Building CXX object src/lib/ec...es/lib__ecl.dir/EKF/mag_fusion.cpp.obj
ninja: build stopped: subcommand failed.
Makefile:155: recipe for target 'px4fmu-v3_default' failed
make: *** [px4fmu-v3_default] Error 1
```

dagar, 2017-10-17:
You need ubuntu_sim_nuttx.sh for building the pixhawk firmware.
It looks you already had the arm compiler installed, but it's a newer version than we currently support.

elbon1, 2017-10-17:
If I have the arm compiler installed, it's from the dev guide since I only installed software from there. I now installed version 5 since the latest is 6, and the same error came up. Do you know what version is required? Thank you

dagar, 2017-10-17:
If you used ubuntu_sim_nuttx.sh it should be gcc 5.4 installed in /opt. Above you mentioned ubuntu_sim.sh, which doesn't install the arm compiler.

Delete the build directories (make clean) and note the C and C++ compiler versions printed at the beginning of cmake configure.

elbon1, 2017-10-17:
Yes my apologies, I actually used ubuntu_sim_nuttx.sh. I checked and I do have gcc 5.4 installed. I deleted the build directories and the compiled versions are outputted below:

```ruby
-- Build Type: MinSizeRel
-- PX4 VERSION: v1.6.5-1518-gc607805
-- CONFIG: nuttx-px4fmu-v3-default
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/local/lib/python2.7/dist-packages/jinja2  
-- C compiler: arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
-- C++ compiler: arm-none-eabi-g++ (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
-- RTPS bridge enabled, send: sensor_combined receive: sensor_baro
-- Configuring 
-- Version: 1.0.6
-- To change the version modify the file configure.ac
-- micro-CDR without check endianness
-- Building and including px4io-v2
-- Using C++03
-- Release build type: MinSizeRel
-- Adding UAVCAN STM32 platform driver
-- NuttX: px4fmu-v3 nsh cortex-m4
-- ROMFS: px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /home/elysiabonello/src/Firmware/build/px4fmu-v3_default
[7/906] Building px4io-v2
ninja: no work to do.
[607/906] Building CXX object src/modu...ient.dir/microRTPS_client_main.cpp.obj
FAILED: /usr/bin/arm-none-eabi-g++   -DBUILD_URI=localhost -DCONFIG_ARCH_BOARD_PX4FMU_V3 -DMODULE_NAME=\"micrortps_client\" -DPX4_MAIN=micrortps_client_app_main -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I. -Isrc -Isrc/modules -I../../src -I../../src/drivers/boards/px4fmu-v3 -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/nuttx/include -INuttX/nuttx/include/cxx -I../../platforms/nuttx/NuttX/apps/include -Iexternal/Install/include -I../../src/modules/micrortps_bridge/micrortps_client -Isrc/modules/micrortps_bridge/micrortps_client/micrortps_client -I../../src/lib/micro-CDR/include -Isrc/lib/micro-CDR/include/microcdr -fno-common -ffunction-sections -fdata-sections -mcpu=cortex-m4 -mthumb -march=armv7e-m -mfpu=fpv4-sp-d16 -mfloat-abi=hard -fno-common -ffunction-sections -fdata-sections -g -fno-exceptions -fno-rtti -std=gnu++11 -fno-threadsafe-statics -DCONFIG_WCHAR_BUILTIN -D__CUSTOM_FILE_IO__ -fcheck-new -Wall -Warray-bounds -Wdisabled-optimization -Werror -Wextra -Wfatal-errors -Wfloat-equal -Wformat-security -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-field-initializers -Wpointer-arith -Wshadow -Wuninitialized -Wunknown-pragmas -Wunused-variable -Wno-unused-parameter -Wunused-but-set-variable -Wformat=1 -Wdouble-promotion -Wno-missing-field-initializers -Wreorder -fvisibility=hidden -include visibility.h -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -Os -DNDEBUG   -Wframe-larger-than=4096 -fno-strict-aliasing -fomit-frame-pointer -funsafe-math-optimizations -ffunction-sections -fdata-sections -fno-strength-reduce -fno-builtin-printf -MMD -MT src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/microRTPS_client_main.cpp.obj -MF src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/microRTPS_client_main.cpp.obj.d -o src/modules/micrortps_bridge/micrortps_client/CMakeFiles/modules__micrortps_bridge__micrortps_client.dir/microRTPS_client_main.cpp.obj -c ../../src/modules/micrortps_bridge/micrortps_client/microRTPS_client_main.cpp
In file included from /usr/arm-none-eabi/include/c++/6.3.1/string:52:0,
                 from src/modules/micrortps_bridge/micrortps_client/micrortps_client/microRTPS_transport.h:35,
                 from ../../src/modules/micrortps_bridge/micrortps_client/microRTPS_client_main.cpp:33:
/usr/arm-none-eabi/include/c++/6.3.1/bits/basic_string.h: In function 'int std::__cxx11::stoi(const wstring&, size_t*, int)':
/usr/arm-none-eabi/include/c++/6.3.1/bits/basic_string.h:5534:42: error: 'wcstol' is not a member of 'std'
   { return __gnu_cxx::__stoa<long, int>(&std::wcstol, "stoi", __str.c_str(),
                                          ^~~
compilation terminated due to -Wfatal-errors.
[607/906] Building CXX object src/modu...rortps_client/microRTPS_client.cpp.obj
FAILED: /usr/bin/arm-none-eabi-g++   -DBUILD_URI=localhost -DCONFIG_ARCH_BOARD_PX4FMU_V3 -DMODULE_NAME=\"micrortps_client\" -DPX4_MAIN=micrortps_client_app_main -D__DF_NUTTX -D__PX4_NUTTX -D__STDC_FORMAT_MACROS -I. -Isrc -Isrc/modules -I../../src -I../../src/drivers/boards/px4fmu-v3 -I../../src/include -I../../src/lib -I../../src/lib/DriverFramework/framework/include -I../../src/lib/matrix -I../../src/modules -I../../src/platforms -INuttX/nuttx/arch/arm/src/armv7-m -INuttX/nuttx/arch/arm/src/chip -INuttX/nuttx/arch/arm/src/common -INuttX/nuttx/include -INuttX/nuttx/include/cxx -I../../platforms/nuttx/NuttX/apps/include -Iexternal/Install/include -I../../src/modules/micrortps_bridge/micrortps_client -Isrc/modules/micrortps_bridge/micrortps_client/micrortps_client -I../../src/lib/micro-CDR/include -Isrc/lib/micro-CDR/include/microcdr -fno-common -ffunction-sections -f [... cut ...]
/usr/arm-none-eabi/include/c++/6.3.1/bits/basic_string.h:5534:42: error: 'wcstol' is not a member of 'std'
   { return __gnu_cxx::__stoa<long, int>(&std::wcstol, "stoi", __str.c_str(),
                                          ^~~
compilation terminated due to -Wfatal-errors.
[607/906] Building CXX object src/lib/...ontrollib.dir/block/BlockParam.cpp.obj
ninja: build stopped: subcommand failed.
Makefile:155: recipe for target 'px4fmu-v3_default' failed
make: *** [px4fmu-v3_default] Error 1

```

elbon1, 2017-10-20:
Starting back from the beginning and the problem is solved. Thanks

dagar, 2017-10-20:
Glad it's working and thanks for the update.

closed 2018-01-01 by LorenzMeier