PX4/PX4-Autopilot #8126: Cannot build Pixhawk firmware

okfynr, 2017-10-13:
I use Ubuntu 14.04  64-bit OS and followed developer guide to make a firmware, but I got such output:
```
dombr@dombr-N56VB:~/src2/Firmware$ make px4fmu-v2_default
CMake Error: The source directory "/home/dombr/src2/Firmware/build/px4fmu-v2_default/-Wno-deprecated" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.
/bin/sh: 1: cd: can't cd to /home/dombr/src2/Firmware/build/px4fmu-v2_default
make: *** [px4fmu-v2_default] Error 2
```
When I toggled commented line
```
CMAKE_ARGS := -Wno-deprecated
```
in ~/src2/Firmware/Makefile file, it resulted in 
```
dombr@dombr-N56VB:~/src2/Firmware$ make px4fmu-v2_default
-- Build Type: MinSizeRel
-- PX4 VERSION: v1.6.5-1479-g577a8ef
-- CONFIG: nuttx-px4fmu-v2-default
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Found PythonInterp: /home/dombr/anaconda3/bin/python (found version "3.5.2") 
-- Found PY_jinja2: /home/dombr/anaconda3/lib/python3.5/site-packages/jinja2  
-- C compiler: arm-none-eabi-gcc (15:5.4.1+svn241155-1) 5.4.1 20160919
-- C++ compiler: arm-none-eabi-g++ (15:5.4.1+svn241155-1) 5.4.1 20160919
-- Building and including px4io-v2
-- Configuring 
-- Version: 1.0.6
-- To change the version modify the file configure.ac
-- micro-CDR without check endianness
-- NuttX: px4fmu-v2 nsh cortex-m4
CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:113 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:143 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:158 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:177 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:182 (add_nuttx_dir)

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:177 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:183 (add_nuttx_dir)

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:177 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:184 (add_nuttx_dir)

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:177 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:185 (add_nuttx_dir)

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:177 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:186 (add_nuttx_dir)

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:177 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:187 (add_nuttx_dir)

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:177 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:188 (add_nuttx_dir)

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:177 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:189 (add_nuttx_dir)

CMake Error at platforms/nuttx/NuttX/CMakeLists.txt:177 (add_dependencies):
  add_dependencies Cannot add target-level dependencies to INTERFACE library
  target "nuttx_build".

Call Stack (most recent call first):
  platforms/nuttx/NuttX/CMakeLists.txt:190 (add_nuttx_dir)

-- ROMFS: px4fmu_common
-- Configuring incomplete, errors occurred!
See also "/home/dombr/src2/Firmware/build/px4fmu-v2_default/CMakeFiles/CMakeOutput.log".
/bin/sh: 1: cd: can't cd to /home/dombr/src2/Firmware/build/px4fmu-v2_default
make: *** [px4fmu-v2_default] Error 2

```
What was done wrong?

r0gelion, 2017-10-16:
Try to update cmake

dagar, 2017-10-17:
We've only been testing Ubuntu 16.04 and newer. We can try to keep it working on older releases, but without CI it's likely to break. @okfynr if upgrading cmake didn't work, and you can't upgrade ubuntu you can build with these docker containers. https://hub.docker.com/r/px4io/px4-dev-nuttx/

sparky002, 2017-11-12:
I got it working successfully by building a later version of cmake on Ubuntu 14.04. The below information I got from this link. I copied the portion that matters so there is no confusion. Credit goes to the poster.

https://askubuntu.com/questions/610291/how-to-install-cmake-3-2-on-ubuntu-14-04

"""""
Compile it yourself

Currently, the latest version is 3.5.2, the following instructions based on this version. You should therefore check the download page and adjust the file name accordingly.

sudo apt-get install build-essential
wget http://www.cmake.org/files/v3.5/cmake-3.5.2.tar.gz
tar xf cmake-3.5.2.tar.gz
cd cmake-3.5.2
./configure
make

Now you can have to choose an installation type to decide, I recommend installing with checkinstall, as this creates a DEB package.

    with checkinstall

    sudo apt-get install checkinstall
    sudo checkinstall

    with make install

    sudo make install

"""""

dagar, 2017-11-12:
Glad to hear you got it working. Going forward we'll try to maintain build environment instructions for the current Ubuntu LTS and previous.

kandai-wata, 2017-11-22:
I also had the same problem with cmake 3.2.2 on Ubuntu 14.04 LTS.
The above solution that @sparky002 had given worked well for me.
I guess cmake below 3.3 has a problem with INTERFACE library, 
so it needs to be upgraded. 

It's written in [stackoverflow](https://stackoverflow.com/questions/35630755/propagate-dependencies-to-header-only-externalproject-with-cmake).

AbhishekGS, 2017-11-29:
I had the same problem with cmake 3.2 on Bash on Windows 10. @sparky002 's solution worked like a charm.

okfynr, 2017-12-03:
I also got it working.
@sparky002 , thanx a lot!

bigbellmercy, 2018-03-20:
Thanks, it works.

closed 2018-07-04 by TSC21