PX4/PX4-Autopilot #8127: Directory structure refactoring

mcharleb, 2017-10-13:
**Feature Request**

One of the criticisms of PX4 is that is difficult to know what code is required for a given platform build. This proposal refactors the PX4 Firmware directory structure to:

- Separate platform specific code into a platform tree
- Aggregate related boards under a "vendor"
- Enable out of tree boards and platforms to be easily integrated into the build system to prevent cloning and forking of the repository and enabling not yet release product development.

This is a RFC to allow anyone with concerns on this change to voice them or to make proposals on how to better address the above issues.

The Directory layout would be the following:
```
../Firmware
├── cmake
│   ├── common
│   ├── templates
│   ├── test
│   └── toolchains
├── Debug
├── Documentation
│   └── px4_hil
│       └── docs
├── integrationtests
│   └── python_src
│       └── px4_it
│           ├── dronekit
│           ├── mavros
│           └── util
├── mavlink
│   └── include
│       └── mavlink
│           ├── v1.0
│           └── v2.0
├── misc
│   └── tones
├── msg
│   ├── ros
│   └── templates
│       ├── px4
│       │   ├── ros
│       │   └── uorb
│       ├── uorb
│       └── urtps
├── platforms
│   ├── nuttx
│   │   ├── cmake
│   │   │   ├── configs
│   │   │   │   └── uavcan_board_ident
│   │   │   ├── nuttx
│   │   │   └── toolchains
│   │   ├── Debug
│   │   ├── Images
│   │   ├── NuttX
│   │   ├── nuttx-patches
│   │   ├── src
│   │   │   ├── drivers
│   │   │   │   ├── batt_smbus
│   │   │   │   ├── bma180
│   │   │   │   ├── bmi055
│   │   │   │   ├── bmi160
│   │   │   │   ├── bmm150
│   │   │   │   ├── bmp280
│   │   │   │   ├── boards
│   │   │   │   │   └── common
│   │   │   │   │       └── stm32
│   │   │   │   ├── bootloaders
│   │   │   │   │   ├── include
│   │   │   │   │   └── src
│   │   │   │   │       ├── arch
│   │   │   │   │       │   └── stm32
[... 580 lines cut ...]
    │           └── drivers
    │               └── boards
    │                   └── s2740vc-v1
    │                       └── bootloader
    └── zubax
        └── nuttx
            ├── boards
            │   └── zubaxgnss-v1
            ├── cmake
            │   └── configs
            │       └── uavcan_board_ident
            ├── nuttx-configs
            │   └── zubaxgnss-v1
            │       ├── bootloader
            │       ├── include
            │       ├── nsh
            │       ├── scripts
            │       └── src
            └── src
                └── drivers
                    └── boards
                        └── zubaxgnss-v1
                            └── bootloader

```

mcharleb, 2017-10-13:
It seems like ROMFS should be broken up. Any suggestions?

mcharleb, 2017-10-13:
Should vendor/px4/nuttx/nuttx-configs/px4-cannode-v1 be vendor/px4/nuttx/nuttx-configs/cannode-v1 and likewise for all the other boards under vendor/px4/nuttx/nuttx-configs? The same for vendor/px4/nuttx/src/drivers/boards?

Nanaijenal85, 2018-07-07:
#9279 Drivers Code Cleanup

dagar, 2019-01-17:
Done!

closed 2019-01-17 by dagar