PX4/PX4-Autopilot #8130: Rpi: Position flight mode seems not functioning well

trigger1996, 2017-10-15:
These days I assembled a quadcopter ,with a Raspi 3B and a custom shield with MP9250, MS5611, PCA9685. This one is really similar to Crossa's.

https://www.baidu.com/link?url=1G6FMZ_7VmyYjRiN1p9Kp0JvSa-zRrrF24e8etGuDXfty5w5bmUMrj2mdhCYw8VtwY1D1pvnuyb6vZwGeGXjgTb7JTsHacSmHbfo4xwpR_G&wd=&eqid=926bdac3000356a70000000659e35873

This one.

Indeed, my copter can fly stable in stablized mode and altitiute mode. However, when switched to Position mode, it cannot stay in or around a small area but starts an oscillation, that is, it flys in a oval-like track, and its radius increases each time it finishs a circle. That's all about the symptom.

My GPS is a ublox M8N, conneted with my Raspi by a USB cable. And I installed Minicom to get USB-TTL driver. In pxh, I input

    _gps start -d /dev/ttyACM0 -i uart -p ubx_

It works. When I input

    _gps status_

I can get correct feedbacks I think.
In the open area, it can find a about 7-9 satellites, however its noise is high, at least 80.

# In all, I wonder what's wrong with my GPS and my copter? What can I do to solove this problem?

P.s.:
1 I've got another copter with Raspi and a Navio shield, I thought its gps is not work well either, and the problem is similar.
2 Sorry about my English

https://logs.px4.io/plot_app?log=788b561d-0cc2-43cb-a3fd-61f2055073d0

dagar, 2017-10-15:
It's an estimator (EKF) issue likely related to your mag and not the GPS.

https://logs.px4.io/plot_app?log=788b561d-0cc2-43cb-a3fd-61f2055073d0#Nav-Estimator-Watchdog

EKF innovation test failures
[image]

[image]

How is your compass mounted? Mount it as far from high current wires as possible.
Then do the full 6 position calibration outdoors away from buildings, vehicles, etc.

trigger1996, 2017-10-15:
Thanks a lot.

My copter's compass is a MPU9250, it funtions as both IMU and compass, so when I assembled my copter, I only take the IMU's orientation into consideration. So, will it have a side effect on it? I'll give it a try tomorrow.

[image]

My MPU9250 is right up the GPS, the blue one.

And about the interference, I have no idea about it,? how about something metal to block my Rpi away from MPU9250? or just an external compass? Both of these I'll give it a try as well.

dagar, 2017-10-15:
Your compass is with the blue GPS on the arm? What's the black piece on the stand? Having the compass mounted away from the frame on a mast is a very good idea.

After that are you sure the compass rotation is correct? With the way it's mounted on the arm isn't the rotation wrong by 45 degrees? Verify the heading in QGC.

trigger1996, 2017-10-16:
My IMU & compass is this one.

[image]

This morning (time zone is different), I recalibrated the compass and rewired the ESC signal cable in the picture above, it did work. My gps works well, what's more, I can spare my hand to take pictures while flying.

Next time when redraw my PCB, I'll consider mounting my compass on a mast like what you suggest.

# Again, thanks.

dagar, 2017-10-16:
Glad to hear it's working now.

closed 2017-10-16 by dagar