PX4/PX4-Autopilot #8141: parameter generation improvements

dagar, 2017-10-17:
For a number of reasons it would be helpful to rename the parameter files to something unique (stop pretending they're .c files) and explicitly list the parameter files per module.

 - rename all parameters files *.c -> *.param or *.px4param
 - include each parameter file as part of the corresponding module definition

**Questions**
Since we'll need to touch every single parameter file do we want to change anything else in the process?
They could simply be xml and then the final param generation would be a merge, or perhaps json would be more user friendly.

**Example: FW_THR_MIN definition**

[image]

**Example: FW_THR_MIN generated xml**

[image]

@hamishwillee @LorenzMeier @mcharleb FYI

bkueng, 2017-10-17:
Duplicate: https://github.com/PX4/Firmware/issues/6115

Instead of JSON or XML, we could also use TOML (as used by the mixer PR in https://github.com/PX4/Firmware/pull/8063). The syntax is simple and pretty much straight-forward: https://github.com/toml-lang/toml#example

And while at it we could add some more meta-data, for example:
- Simple/Advanced flag (probably not to easy to decide though)
- internal flag: parameters that should not be exposed (or modified) via UI, such as temperature calibration params or total flight time.

dagar, 2017-10-17:
+1 for internal flag (or similar). Giving these special treatment will improve several different areas.

One factor is if we want to preserve the parameters.xml and existing .px4 format, or still ultimately generate the same thing for compatibility. 

We should also consider if we want additional types. uint64, double, bool, etc 
I'm aware of the implications for the PX4 param library.

closed 2018-02-21 by PX4BuildBot