PX4/PX4-Autopilot #8142: px4fmu driver refactor

dagar, 2017-10-17:
Notes and ideas for a px4fmu refactor.

Split drivers/px4fmu into the following separate modules
 - px4fmu (PWM output + overall PWM/GPIO configuration)
 - rc_input (runs in work queue at X00Hz) (could eventually be merged with drivers/spektrum_rc, drivers/ocpoc_sbus_rc_in, and likely others)
 - safety_button module (runs in work queue at 10Hz)
 - i2c command (only bus configuration, but could be extended to scan bus, etc)
 - spi command (only a reset currently, but could be extended to scan bus, etc)
 - Anything else that can be deleted or moved to a separate module? GPIO IOCTL configuration? PWM capture?

The remaining px4fmu core should be extended to support multiple instances (https://github.com/PX4/Firmware/pull/7208).

This should help save some space on platforms that don't use the other pieces. (https://github.com/PX4/Firmware/pull/7993) @zehortigoza FYI

https://github.com/dagar/Firmware/tree/pr-fmu_split

@iZhangHui @davids5 @rolandash

iZhangHui, 2017-10-18:
@dagar  Great work! Thank you. Your PR makes px4fmu codes more clean, clear and maintenance easier.
At the same, it will reduce firmware size for some flight controllers and ram consumption.
One concern is that ModuleBase doesn't support modules which allow multiple instances.

dagar, 2017-10-19:
We can either extend it or make a new version that does. It's something we'll need to do for an improved driver framework either way.

There are still lots of inconsistencies in px4fmu when used as MAIN vs AUX that need to be fixed.

closed 2018-02-21 by PX4BuildBot